CODeRUS/GyverLampCpp #1: There are some compile problems.

phplego, 2019-09-06:
I ran compilation using the sources w/o any modifications

```
> Executing task: platformio run <

Processing nodemcu (framework: arduino; platform: espressif8266@2.2.2; board: nodemcuv2)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 2.2.2 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: toolchain-xtensa 2.40802.190218 (4.8.2), tool-esptool 1.413.0 (4.13), tool-esptoolpy 1.20600.0 (2.6.0), framework-arduinoespressif8266 2.20502.0 (2.5.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
|-- <ESP Async WebServer> 1.2.2
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ArduinoJson> 6.12.0
|-- <ESPAsyncWiFiManager> 0.21 #7f37678
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP Async WebServer> 1.2.2
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ArduinoJson> 6.12.0
|   |-- <DNSServer> 1.1.1
|   |   |-- <ESP8266WiFi> 1.0
|-- <FastLED> 3.3.2
|   |-- <SPI> 1.0
|   |-- <EspSoftwareSerial> 5.0.4
|-- <FastLED NeoMatrix> 1.0 #0daa5c3
|   |-- <FastLED> 3.3.2
|   |   |-- <SPI> 1.0
|   |   |-- <EspSoftwareSerial> 5.0.4
|   |-- <Adafruit GFX Library> 1.5.6
|   |   |-- <SPI> 1.0
|   |-- <Framebuffer GFX> 1.0 #a5336ed
|   |   |-- <FastLED> 3.3.2
|   |   |   |-- <SPI> 1.0
|   |   |   |-- <EspSoftwareSerial> 5.0.4
|   |   |-- <Adafruit GFX Library> 1.5.6
|   |   |   |-- <SPI> 1.0
|-- <Framebuffer GFX> 1.0 #a5336ed
|   |-- <FastLED> 3.3.2
|   |   |-- <SPI> 1.0
|   |   |-- <EspSoftwareSerial> 5.0.4
|   |-- <Adafruit GFX Library> 1.5.6
|   |   |-- <SPI> 1.0
|-- <Adafruit GFX Library> 1.5.6
|   |-- <SPI> 1.0
|-- <NTPClient> 3.1.0
|-- <ArduinoJson> 6.12.0
|-- <arduinoFFT> 1.5
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <GyverButton>
|-- <EEPROM> 1.0
Compiling .pio/build/nodemcu/libfd3/FastLED_ID126/FastLED.cpp.o
Compiling .pio/build/nodemcu/libfd3/FastLED_ID126/bitswap.cpp.o
Compiling .pio/build/nodemcu/libfd3/FastLED_ID126/colorpalettes.cpp.o
Compiling .pio/build/nodemcu/libfd3/FastLED_ID126/colorutils.cpp.o
[... 85 lines cut ...]
src/Settings.cpp:226:53: error: no matching function for call to 'fs::FS::open(const char*&)'
         File settings = SPIFFS.open(settingsFileName);
                                                     ^
src/Settings.cpp:226:53: note: candidates are:
In file included from src/Settings.cpp:9:0:
/home/dub/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.h:190:10: note: fs::File fs::FS::open(const char*, const char*)
     File open(const char* path, const char* mode);
          ^
/home/dub/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.h:190:10: note:   candidate expects 2 arguments, 1 provided
/home/dub/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.h:191:10: note: fs::File fs::FS::open(const String&, const char*)
     File open(const String& path, const char* mode);
          ^
/home/dub/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.h:191:10: note:   candidate expects 2 arguments, 1 provided
*** [.pio/build/nodemcu/src/Settings.cpp.o] Error 1
======================================================================== [FAILED] Took 3.42 seconds ========================================================================

Environment    Status    Duration
-------------  --------  ------------
sonoff         IGNORED
nodemcu        FAILED    00:00:03.421
esp32dev       IGNORED
=================================================================== 1 failed, 0 succeeded in 00:00:03.421 ===================================================================
```

Could anybody say what I do wrong? Thanks

phplego, 2019-09-06:
By the way, thank you for this PlatformIO adaptation! I was really impressed the way you splitted that Gyver's mesh to clean and simple classes ^-^

CODeRUS, 2019-09-06:
Thanks! I didnt testes esp8266 build for a long time. Fixed by 2cffa3d8e2986a13227e25a980571be451a4acbb

CODeRUS, 2019-09-06:
> 
> 
> By the way, thank you for this PlatformIO adaptation! I was really impressed the way you splitted that Gyver's mesh to clean and simple classes ^-^

It took me some time before i can actually use it

closed 2019-09-06 by CODeRUS

phplego, 2019-09-06:
It works!
[image]

CODeRUS, 2019-09-06:
dont forget to upload spiffs bbefore firing it up ;)

CODeRUS, 2019-09-06:
also wifi configuration page is availabble at /wifi endpoint

Some day i will clean things up and write proper manual of using it :D

phplego, 2019-09-06:
> dont forget to upload spiffs bbefore firing it up ;)

Yes yes, ive uploaded. It works!

It shows trianangle running pixels and then diagonal rainbow. After that esp goes to restart

```
Happy debugging!
SPIFFS Settings file exists: true
SPIFFS Read effect Sparkles, speed: 121, scale: 12, brightness: 255
SPIFFS Read effect Fire, speed: 37, scale: 1, brightness: 80
SPIFFS Read effect Vertical rainbow, speed: 100, scale: 18, brightness: 80
SPIFFS Read effect Horizontal rainbow, speed: 100, scale: 18, brightness: 80
SPIFFS Read effect Color change, speed: 112, scale: 1, brightness: 80
SPIFFS Read effect Madness 3D, speed: 50, scale: 100, brightness: 80
SPIFFS Read effect Clouds 3D, speed: 1, scale: 100, brightness: 80
SPIFFS Read effect Lava 3D, speed: 1, scale: 100, brightness: 80
SPIFFS Read effect Plasma 3D, speed: 1, scale: 100, brightness: 80
SPIFFS Read effect Rainbow 3D, speed: 1, scale: 52, brightness: 80
SPIFFS Read effect Rainbow stripe 3D, speed: 1, scale: 47, brightness: 80
SPIFFS Read effect Zebra 3D, speed: 1, scale: 26, brightness: 80
SPIFFS Read effect Forest 3D, speed: 1, scale: 64, brightness: 80
SPIFFS Read effect Ocean 3D, speed: 1, scale: 24, brightness: 80
SPIFFS Read effect Single color, speed: 10, scale: 1, brightness: 80
SPIFFS Read effect Snow, speed: 255, scale: 33, brightness: 80
SPIFFS Read effect Matrix, speed: 255, scale: 25, brightness: 80
SPIFFS Read effect Lighters, speed: 127, scale: 10, brightness: 80
SPIFFS Read effect Clock vertical, speed: 250, scale: 1, brightness: 80
SPIFFS Read effect Clock horizontal colored, speed: 250, scale: 1, brightness: 80
SPIFFS Read effect Clock horizontal, speed: 250, scale: 1, brightness: 80
SPIFFS Read effect Clock horizontal single, speed: 250, scale: 1, brightness: 80
SPIFFS Read effect Sound spectrometer, speed: 1, scale: 1, brightness: 236
SPIFFS Read effect Starfall, speed: 80, scale: 100, brightness: 80
SPIFFS Read effect DiagonalRainbow, speed: 100, scale: 20, brightness: 80
SPIFFS Read effect Sparkles, speed: 121, scale: 12, brightness: 255
SPIFFS Read effect Fire, speed: 37, scale: 1, brightness: 80
SPIFFS Read effect Vertical rainbow, speed: 100, scale: 18, brightness: 80
SPIFFS Read effect Horizontal rainbow, speed: 100, scale: 18, brightness: 80
SPIFFS Read effect Color change, speed: 112, scale: 1, brightness: 80
SPIFFS Read effect Madness 3D, speed: 50, scale: 100, brightness: 80
SPIFFS Read effect Clouds 3D, speed: 1, scale: 100, brightness: 80
SPIFFS Read effect Lava 3D, speed: 1, scale: 100, brightness: 80
SPIFFS Read effect Plasma 3D, speed: 1, scale: 100, brightness: 80
SPIFFS Read effect Rainbow 3D, speed: 1, scale: 52, brightness: 80
SPIFFS Read effect Rainbow stripe 3D, speed: 1, scale: 47, brightness: 80
SPIFFS Read effect Zebra 3D, speed: 1, scale: 26, brightness: 80
SPIFFS Read effect Forest 3D, speed: 1, scale: 64, brightness: 80
SPIFFS Read effect Ocean 3D, speed: 1, scale: 24, brightness: 80
SPIFFS Read effect Single color, speed: 10, scale: 1, brightness: 80
SPIFFS Read effect Snow, speed: 255, scale: 33, brightness: 80
SPIFFS Read effect Matrix, speed: 255, scale: 25, brightness: 80
SPIFFS Read effect Lighters, speed: 127, scale: 10, brightness: 80
SPIFFS Read effect Clock vertical, speed: 250, scale: 1, brightness: 80
SPIFFS Read effect Clock horizontal colored, speed: 250, scale: 1, brightness: 80
SPIFFS Read effect Clock horizontal, speed: 250, scale: 1, brightness: 80
SPIFFS Read effect Clock horizontal single, speed: 250, scale: 1, brightness: 80
SPIFFS Read effect Sound spectrometer, speed: 1, scale: 1, brightness: 236
SPIFFS Read effect Starfall, speed: 80, scale: 100, brightness: 80
SPIFFS Read effect DiagonalRainbow, speed: 100, scale: 20, brightness: 80
DiagonalRainbow activated!

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
```

(by the way I changed led pin to 5 (it was 6))

CODeRUS, 2019-09-06:
will try to check what is wrong here

phplego, 2019-09-06:
hehe, I cleared settings.json and it stoped to restart. Looks like everything works as expected!

CODeRUS, 2019-09-06:
works after reboot also?

phplego, 2019-09-06:
Yes, effects are working, web-panel also. But only if I comment some effects in the EffectManager. Playing around enabling/disabling different effects lead me to an idea that this is a memory problem. Because it does not mean which are disabled, just a number of effects. Out of memory maybe. Will check it later..

CODeRUS, 2019-09-06:
check after e97d9907e28eb602933ee7c920f22b918545a143

phplego, 2019-09-06:
Updated. Compiled. Uploaded spiffs image.

It works. But if I enable all the effects then web-panel never loads, infinite pending. If i disable several effects (eg last 13) then in loads normally (can change effect, ajust speed and etc)

[image]

I will print out free memory, i think it become clear is it memory problem or not..

phplego, 2019-09-06:
Yes, looks like it is!
When disable ~half of effects then it is about 30 kb free heap. And when I open web-panel it dive down to ~18kb free heap.

When all the effects are enabled it is about 15 kb free heap. And I guess it isn't enough to process web server request

CODeRUS, 2019-09-06:
Check https://aliexpress.com/item/32817106037.html this is the cheapest esp32 with unsoldered pins i found. works flaweless :D

phplego, 2019-09-06:
> Check https://aliexpress.com/item/32817106037.html this is the cheapest esp32 with unsoldered pins i found. works flaweless :D

yea, thanks! it is really cheap. I think I buy it. Anyway I will try to reduce memory consumption. Since esp8266 is declared in platformio.ini.

I beleive allocate memory for arrays and etc dynamically may help.. will see.. interesting challenge anyway

CODeRUS, 2019-09-06:
sure, i will look at it again later when new led matrix modules will arrive

phplego, 2019-09-06:
Moving allocate arrays from constructor of **NoiseEffect** to _activate()_ and freeing memory on _deactivate()_ solved my memory problem 

[image]

CODeRUS, 2019-09-07:
good job! please check b078028e47b3a1ef0b6e80b225702e90fc9a9d54

phplego, 2019-09-07:
> good job! please check [b078028](https://github.com/CODeRUS/GyverLampCpp/commit/b078028e47b3a1ef0b6e80b225702e90fc9a9d54)

yes, it reduced memory consumption even more. Awesome!

phplego, 2019-09-07:
I have another question about SoundEffect. I checked it by connecting generator built in my multimeter to A0 pin, and looks like it works. But what optimal levels are? Maybe you can share the scheme how to connect it, for example, to line audio output (https://en.wikipedia.org/wiki/Line_level) ? Thanks in advance

CODeRUS, 2019-09-07:
I used it with KY-037
https://s.click.aliexpress.com/e/cTXadrR8

CODeRUS, 2019-09-07:
for stereo spectrometer i'm using this scheme: https://github.com/G6EJD/ESP32-8-Octave-Audio-Spectrum-Display/blob/master/Stereo_Input_Wiring.jpg

110k ohm resistors on both sides, 47nF capacitors.

arrowcircle, 2019-12-27:
@CODeRUS Hey! Thanks for the great work rewriting the firmware to PlatformIO.
I got similar problems with compilation with latest master.

Here is the log from platformio

```txt
In file included from src\LampWebServer.cpp:14:0:
.pio\libdeps\nodemcu\ESP Async WebServer\src/ESPAsyncWebServer.h:36:25: fatal error: ESPAsyncTCP.h: No such file or directory

*********************************************************************
* Looking for ESPAsyncTCP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPAsyncTCP.h"
* Web  > https://platformio.org/lib/search?query=header:ESPAsyncTCP.h
*
*********************************************************************

 #include <ESPAsyncTCP.h>
                         ^
compilation terminated.
In file included from src\Settings.cpp:6:0:
.pio\libdeps\nodemcu\ESP Async WebServer\src/ESPAsyncWebServer.h:36:25: fatal error: ESPAsyncTCP.h: No such file or directory

*********************************************************************
* Looking for ESPAsyncTCP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPAsyncTCP.h"
* Web  > https://platformio.org/lib/search?query=header:ESPAsyncTCP.h
*
*********************************************************************

 #include <ESPAsyncTCP.h>
                         ^
compilation terminated.
*** [.pio\build\nodemcu\src\LampWebServer.cpp.o] Error 1
*** [.pio\build\nodemcu\src\Settings.cpp.o] Error 1
In file included from src/MyMatrix.h:13:0,
                 from src/effects/Effect.h:2,
                 from src\effects\basic\ClockHorizontal1Effect.h:2,
                 from src\effects\basic\ClockHorizontal1Effect.cpp:1:
.pio\libdeps\nodemcu\FastLED_ID126/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.002
 #    pragma message "FastLED version 3.003.002"
                     ^
In file included from src/MyMatrix.h:13:0,
                 from src/effects/Effect.h:2,
                 from src\effects\basic\ClockEffect.h:2,
                 from src\effects\basic\ClockEffect.cpp:1:
.pio\libdeps\nodemcu\FastLED_ID126/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.002
 #    pragma message "FastLED version 3.003.002"
                     ^
In file included from src/MyMatrix.h:13:0,
                 from src/effects/Effect.h:2,
                 from src\effects\basic\ClockHorizontal3Effect.h:2,
                 from src\effects\basic\ClockHorizontal3Effect.cpp:1:
.pio\libdeps\nodemcu\FastLED_ID126/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.002
 #    pragma message "FastLED version 3.003.002"
                     ^
In file included from src/MyMatrix.h:13:0,
                 from src/effects/Effect.h:2,
                 from src\effects\basic\ClockHorizontal2Effect.h:2,
                 from src\effects\basic\ClockHorizontal2Effect.cpp:1:
.pio\libdeps\nodemcu\FastLED_ID126/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.002
 #    pragma message "FastLED version 3.003.002"
                     ^
In file included from .pio\libdeps\nodemcu\FastLED_ID126/FastLED.h:65:0,
                 from src/MyMatrix.h:13,
                 from src/effects/Effect.h:2,
                 from src\effects\basic\ClockHorizontal1Effect.h:2,
                 from src\effects\basic\ClockHorizontal1Effect.cpp:1:
.pio\libdeps\nodemcu\FastLED_ID126/fastspi.h:130:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
[... 159 lines cut ...]
                 from src/effects/Effect.h:2,
                 from src\effects\basic\ClockHorizontal1Effect.h:2,
                 from src\effects\basic\ClockHorizontal1Effect.cpp:1:
.pio\libdeps\nodemcu\FastLED_ID126/fastspi.h:130:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
In file included from src/MyMatrix.h:13:0,
                 from src/effects/Effect.h:2,
                 from src\effects\basic\ClockHorizontal2Effect.h:2,
                 from src\effects\basic\ClockHorizontal2Effect.cpp:1:
.pio\libdeps\nodemcu\FastLED_ID126/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.002
 #    pragma message "FastLED version 3.003.002"
                     ^
In file included from .pio\libdeps\nodemcu\FastLED_ID126/FastLED.h:65:0,
                 from src/MyMatrix.h:13,
                 from src/effects/Effect.h:2,
                 from src\effects\basic\ClockHorizontal2Effect.h:2,
                 from src\effects\basic\ClockHorizontal2Effect.cpp:1:
.pio\libdeps\nodemcu\FastLED_ID126/fastspi.h:130:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
```

Is it possible, that  versions/commits of dependencies are not fixed and there were major changes in the API?

Any ideas how to fix this?

CODeRUS, 2019-12-27:
try adding AsyncTCP to common.lib_deps

arrowcircle, 2019-12-27:
@CODeRUS Thats not the problem. I added it and got new errors (in second log file).

CODeRUS, 2019-12-27:
repalce setFinishedCallback with onFinished, setNotFoundCallback with onNotFound

seems forgot to push something :) will check next week

CODeRUS, 2020-01-03:
pushed forgotten changes to git, thanks for noticing