CODeRUS/GyverLampCpp #2: WiFi Issues with ESP8266

arrowcircle, 2019-12-28:
Hey!
I found few problems with this version of code.

First thing is that code does not start wifi manager according to logs and start in AP mode. Here is the log output of device start up:

```txt
Happy debugging!
Flash ide  size: 4194304 bytes
Flash ide speed: 40000000 Hz
Flash ide mode:  DIO
Flash real id:   00164020
Flash real size: 4194304 bytes

Flash Chip configuration ok.
Sketch size: 531536
Sketch free: 2613248
FreeHeap: 40936
Initializing EffectsManager
Initializing Settings
SPIFFS Settings file exists: true
SPIFFS Settings file size: 2868
Set max brightness to: 80
Set current limit to: 1000
Set rotation to: 3
Framebuffer_GFX::begin Width: 16 Height: 16 Num Pixels: 256
Initializing web server at: 80
AutoConnect started

AP IP address:
8.8.8.8
Starting DNS server: success
AutoConnect finished
mDNS responder (firelamp) started!
Announcing http (tcp) service on port 80
Setup mode entered. No effects!
Scan done
Found networks: 8
```
If I upload current master version of the code (with corrections of the callback), I cant connect to AP in any way. I tried changing memory profile of LWIP, but it does not help. When I try to connect to the device from the phone or notebook it does not show anything in the terminal and connection just hangs. Logs are empty too

I tried to update platform version to espressif8266@2.3.2. This requires to fix U_SPIFFS to U_FS somewhere in the code.

After that I can connect to the hotspot of the lamp, but there are a lot of problems:

1. Captive portal does not close on both phone and notebook (apple)
2. WS connection does not work, "lamp" in the interface is always red, inspector shows connection error for websockets.

Led functionality works great, but I cant do anything with WiFi.

Any ideas how to fix this?

CODeRUS, 2020-01-03:
Updated to make compatible with 2.3.2 core

CODeRUS, 2020-01-03:
Make sure you built web ui, react ui aand uploaded fs to module

CODeRUS, 2020-01-03:
You can skip previous step and download data_src from https://github.com/CODeRUS/GyverLampCpp/releases and unpack it to project folder

CODeRUS, 2020-01-03:
then execute `platformio run --target uploadfs -e nodemcu`

arrowcircle, 2020-01-03:
Thanks for the fixes, but there are still problems:

1. Master branch still miss `ESPAsyncTCP` dependency.
2. I still cant access wifimanager page to connect to local wifi. I do see captive portal with turn on/off button, but it does not work.
3. Websockets connection does not work. I uploaded FS twice.

Any ideas how to fix it?

CODeRUS, 2020-01-03:
Try opening /wifi or /wifi.html

arrowcircle, 2020-01-03:
> Try opening /wifi or /wifi.html

I cant, there is no URL prompt in captive portal window :(

arrowcircle, 2020-01-03:
Also, there are .pro files with some hardcoded paths like

```
INCLUDEPATH += "$${HOMEDIR}/workplace/GyverLamp/firmware/GyverLampCpp/include"
INCLUDEPATH += "$${HOMEDIR}/workplace/GyverLamp/firmware/GyverLampCpp/src"
INCLUDEPATH += "$${HOMEDIR}/workplace/GyverLamp/firmware/GyverLampCpp/lib/GyverButton" 
```

Is it relevant in any way?

CODeRUS, 2020-01-03:
ah i found small problem in my ci script :D
https://github.com/CODeRUS/GyverLampCpp/commit/0f70a9178b57cda16d266adb5e3021d9e2ee9b4e

plz grab new artifacts from github releases =)

also added that missing dependency to list

arrowcircle, 2020-01-03:
@CODeRUS Now I can connect to WiFi properly. But there are no tab to change wifi settings in the menu. Is it possible to add it?

CODeRUS, 2020-01-03:
it is planned, but no eta. you can always open /wifi page when needed

closed 2020-01-03 by arrowcircle