CODeRUS/GyverLampCpp #3: Compile error in data_compress.py

kagjerde, 2020-02-08:
Hi, thanks for a wery good implementation, everything works, but I have not managed to comile the FS myself. I have ony used the comitted data_src catalogue and this works, but gives an error if you use Python (I guess 3.5 or higer) 3.8.1, when running data_compress.py - it gives the error: "TypeError: memoryview: a bytes-like object is required, not 'str'".

After some testing i replaced 'wb' with 'wt', in (data_compress.py)
line 55. "with open(file) as src, gzip.open(os.path.join(data_dir, os.path.basename(file) + '.gz'), 'wt') as dst:".

Now it compiles and the file system works, by uploading using:
"platformio run --target uploadfs -e nodemcu" in the terminal as stated in an earlier issue report.

Another small issue to be aware of is when uploading the configuration without setting the MQTT IP (in the config file), the nodemcu (I assue it's the same for other targets) reboots when you try to change effects. A better implementation would be to set the MQTT IP in the WEB GUI.

I like the Home Assistant integration, as this gives the possibility to show different effects like rain - according to the weather forecast or any other events.

I miss two things in the WEB GUI:
- Possibility to set MQTT IP
- Change WIFI settings

Other feature requests are more effects linked to events in Home Assistant like:
- Weather: Sun, Rain, Snow, Overcast, Lightning, temperature (like a mercury scale) etc
- Events: Arrived home, Light on (just blink - and then go back to the previous setting), alarm (police blue rotating flash), fire alarm (red-rotating flash). etc.
- Media: Next song, volume state, etc

I will try to implement some of these, but my programming skills is a bit limitted.

CODeRUS, 2020-02-08:
Thanks for reporting. I'll take a look at data compress, or you can create PR yourself :)

For the settings update you can simply upload new json via upgrade dialog, that's why i don't bother making any settings ui.

arrowcircle, 2020-02-10:
Maybe all compression should be made with webpack?

CODeRUS, 2020-02-10:
i have no idea what is webpack 🙄

arrowcircle, 2020-02-14:
@CODeRUS it it **the way** how modern frontend is built

CODeRUS, 2020-02-14:
@arrowcircle cool! cn you make a pr for that?

top4ek, 2020-02-25:
Thanks for "wt"-fix :) Spent 2 hours reinstalling platformio and cleaning system from python2.

CODeRUS, 2020-03-15:
Fixed by compressing static dduring build

closed 2020-03-15 by CODeRUS