CODeRUS/GyverLampCpp #11: Noise effects are segfaulting

top4ek, 2020-03-15:
17x55 matrix fails to render all nouse effects.

CODeRUS, 2020-03-15:
could be not enough heap for 3d arrays

CODeRUS, 2020-03-15:
uncomment last lines in main.cpp and check log

top4ek, 2020-03-17:
It's about 23k free before turning on noise effect. 

If i'm right, noise effects requires 4 * 17 * 55 = 3740 bytes, or i miss smth?

CODeRUS, 2020-03-17:
no idea actually :)

you can try to print heap right inside constructor, if it crashing soon

top4ek, 2020-03-17:
Hm, it's not in activation function.

```
<< {"event":"ACTIVE_EFFECT","data":5}
**Activate FreeHeap: 22456**
**Generated arrays FreeHeap: 20912**
Sending state
{
  "state": "ON",
  "brightness": 255,
  "speed": 50,
  "scale": 100,
  "effect": "Madness 3D"
}
State sent: success

Exception (3):
epc1=0x40100d6c epc2=0x00000000 epc3=0x00000000 excvaddr=0x40030801 depc=0x00000000
```

CODeRUS, 2020-03-17:
put in fillNoise8 as well

top4ek, 2020-03-17:
It doesn't hit it :(

CODeRUS, 2020-03-17:
fixed

closed 2020-03-17 by CODeRUS

top4ek, 2020-03-18:
Works like a charm :) 
Have you tried ESP.wdtDisable()?

CODeRUS, 2020-03-18:
if disable soft watchdog on esp8266 it will hang at hw one.

see: https://www.sigmdel.ca/michel/program/esp8266/arduino/watchdogs_en.html

commit 3ad252bc (2020-03-17): [noise] Fixed noise effects for big matrix and slow chips

Closes #11