CODeRUS/GyverLampCpp #12: how to maually build FS?

kagjerde, 2020-03-27:
I have gor everyting to work and have now tried to add some effects.
If I just change the effects withour renaming or adding a new effect it works, but when I add a new one and update the souce code (not FS) it works for about a minutte. And then it restarts.
The log states :
"
Exception (29):
epc1=0x40208a95 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000"

I do not know why it fails, but I suspect that it is related to memory allocation and the fact that I have not updated the FS files (except settings.json that I have updated).

I have tried to follow the instructions to manually build the new structure:
build https://github.com/CODeRUS/led-lamp-webui
build https://github.com/CODeRUS/wifimanager-react-page
But I have not been able to get it working (I am new to yarn). I am using VSC.

Can you please comment if the failure is likely to be caused by this? and if possible give a quick instruction how to build the FS manually?
Some questions:
- Whent packages in VSC do I need to install?
- Command line instructions to build (step by step if possible)

Thanks in advance.

CODeRUS, 2020-03-27:
for web stuff it should buildd just fine with latest node and just running in terminal `npm install && npm run build` for each folder (see https://github.com/CODeRUS/GyverLampCpp/blob/master/.travis.yml#L27) then grab conmpressed artifacts to data folder

CODeRUS, 2020-03-27:
Make sure you have correct names of effects mapped to json.

closed 2020-04-26 by CODeRUS