CODeRUS/GyverLampCpp #16: /effects returns null

top4ek, 2020-05-06:
Just updated to the last firmware ⅔ of my lamps and noticed that they became unstable. Sometimes i see in logs problems with MQTT connections and sometimes, when i try to load web interface it's just segfaults. Can't tell smth more specific except /effects returns "null" instead of mySettings contents and and prettyJson returns the same.

effectJson shows up with current mode and effect list is updated in MQTT, I can swith them via home assistant.

All this happens with nodemcu.

CODeRUS, 2020-05-07:
oh, effects json is dramatically big these times, and esp8266 heap size is so small... i'll check what can i do here

top4ek, 2020-05-07:
Well, that's exmplain why lamp doesn't boots after power restart and settings are gone :)
 
Some time ago i thought about splitting settings.json to something like effects.json, network.json and matrix.json or even generate effects.json on the fly(i'm not sure if it possible with cpp in easy way).

CODeRUS, 2020-05-07:
this is exactly what /effects enpoint is for

CODeRUS, 2020-05-07:
but not enough heap to generate this json on the fly :D

top4ek, 2020-05-07:
But it's just generates json from mySettings which gets info from settings.json. I meant to get somehow list of effects from compiled classes(but it the way of interperetators). BTW, why not just transfer settings.json as all other assets?

CODeRUS, 2020-05-07:
get from /settings.json was in previous commits of webui, but it has nothing to help your problem. no heap to generate new settings.json with new settings as well :D

CODeRUS, 2020-05-07:
please check latest master, did some stupid optimizations here.
no ideas for better optimizations so far

top4ek, 2020-05-09:
Looks like it helped :)

Now I'm afraid it will not work with 17x60 matrix :)

CODeRUS, 2020-05-09:
let me know of your success

closed 2020-05-09 by CODeRUS