CODeRUS/GyverLampCpp #17: MQTT settings lost while saving settings.json

top4ek, 2020-05-09:
It's me again with a new bug. :)
Saving settings.json to spiffs goes without mqtt section. So after reboot lamp doesn't establish connection to it. (1. upload correct spiffs. 2. Change smth, like brightness. 3. wait for a while. 4. check settings.json).

BTW: I don't get why we need effects.json and same data in settings.json?

CODeRUS, 2020-05-09:
seems i failed most important commit ever :D
should be fixed now

top4ek, 2020-05-10:
Thanks.
Only one thing fix settings.json in assets repo to:
```
{
  "activeEffect":1,
  "working":true,
  "matrix":{
    "width":16,
    "height":16,
    "segments":1,
    "type":15,
    "maxBrightness":255,
    "currentLimit":3000,
    "rotation":2
  },

  "connection": {
    "mdns":"firelamp",
    "apName":"Fire Lamp",
    "ntpServer":"europe.pool.ntp.org",
    "ntpOffset":10800},
  "mqtt":{
      "host":"",
      "port":1883,
      "username":"",
      "password":""},
    "spectrometer":{
      "active":false
    }
  }
}
```
Generally it works, will see :)

top4ek, 2020-05-10:
Spectrometer effects are missing. :)

CODeRUS, 2020-05-10:
You using spectrometer?

top4ek, 2020-05-10:
Not now, waiting for mic :)

CODeRUS, 2020-05-10:
Commited stuff

top4ek, 2020-05-10:
Can we use Settings to check if spectrometer is enabled [here](https://github.com/CODeRUS/GyverLampCpp/commit/5d1b22a4095622b4fdb5c58bb7bb980f73f57923#diff-0a34c3418063465b2c76785510178ba5R286)?

CODeRUS, 2020-05-10:
Probably. I'll review spectrometer stuff later. But FYI commented stuff does not compile and a little more free space in heap ;)

closed 2020-05-10 by top4ek