CODeRUS/GyverLampCpp #19: Boot Loop on NodeMCU

fitim7, 2020-05-25:
If flashing this on a NodeMCU (ESP-12E) I'm always getting this:

`
Happy debugging!
Flash ide  size: 4194304 bytes
Flash ide speed: 40000000 Hz
Flash ide mode:  DIO
Flash real id:   0016405E
Flash real size: 4194304 bytes

Flash Chip configuration ok.
Sketch size: 610048
Sketch free: 2535424
FreeHeap: 38792
Initializing EffectsManager
Initializing Settings
SPIFFS Settings file exists: true
SPIFFS Settings file size: 455
SPIFFS Effects file exists: false
Initializing MyMatrix
Set max brightness to: 80
Set current limit to: 1000
Set rotation to: 3
Framebuffer_GFX::begin Width: 16 Height: 16 Num Pixels: 256
Initializing web server at: 80
AutoConnect started

AP IP address: 
8.8.8.8
Starting DNS server: success
AutoConnect finished
Initializing LocalDNS
mDNS responder (firelamp) started!
Announcing http (tcp) service on port 80

Exception (28):
epc1=0x40202324 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffc80 end: 3fffffc0 offset: 01a0
3ffffe20:  3fff4e84 3fff1120 3fff30fc 4021825b  
3ffffe30:  00706300 00000000 80100776 70747400  
3ffffe40:  00000000 80000001 3fff4e2c 40227fcd  
3ffffe50:  3fff2584 00000001 00000001 402033c6  
3ffffe60:  3fff2584 00000001 00000001 4022a194  
3ffffe70:  00000000 3fff1120 80fea172 4023b8f0  
3ffffe80:  08080808 3fff0035 3ffea172 4022be55  
3ffffe90:  00000000 00000007 3fff1120 4022c0d4  
3ffffea0:  08080808 00ffffff 08080808 00000001  
3ffffeb0:  3fff557c 3fff1004 3fff1120 4022a3ce  
3ffffec0:  4023b8f0 08080808 4023b8f0 08080808  
3ffffed0:  4023b8f0 08080808 3fffff10 4022d31c  
3ffffee0:  3fffff50 00000001 3fffff10 4022d34c  
3ffffef0:  3fffff50 00000000 00000000 3fff48e4  
3fffff00:  3fffff50 00000001 3fff557c 4022a429  
3fffff10:  65726946 6d614c20 89ff0070 4022d300  
3fffff20:  3fff0ab4 80ff557c 3fffff00 4022d34c  
3fffff30:  80ff4904 3fff5500 40238978 80238964  
3fffff40:  3fff0ab4 3fff557c 3fff557c 40203693  
3fffff50:  65726946 6d614c20 89ff0070 4022c0d4  
3fffff60:  3fff4e0c 00000000 3fff1120 3fff1000  
3fffff70:  3fff4e0c 00000000 3fff1120 40218450  
3fffff80:  00000001 00000004 feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3fff126c  
3fffffa0:  3fffdad0 00000000 3fff122c 4022e324  
3fffffb0:  feefeffe feefeffe 3ffe862c 401005ed  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld

Happy debugging!
Flash ide  size: 4194304 bytes
Flash ide speed: 40000000 Hz
Flash ide mode:  DIO
Flash real id:   0016405E
Flash real size: 4194304 bytes
....
`

What is wrong here? Btw, thanks for your effort in this project!

CODeRUS, 2020-05-25:
oh it was ages since i tried hotspot mode
i'll check what is wrong :D

fitim7, 2020-05-25:
How do you use it then? I just build and uploaded your newest release. In settings I didn't find option for Ssid and password. How do you connect then the Espresso to wifi :D

CODeRUS, 2020-05-25:
oh, seems you missing effects.json file
build and flash new spiffs

and all my devices already remembered ssid to connect and never asks for reconnection, but this is not the problem anyways ;)

fitim7, 2020-05-25:
Sorry, but Im new to this. I just grabbed your latest data.zip. Then open this in platform.io and then:
1. Erase flash
2. Build
3. Upload
4. Upload File System image

Now im getting this:
```
Happy debugging!
Flash ide  size: 4194304 bytes
Flash ide speed: 40000000 Hz
Flash ide mode:  DIO
Flash real id:   0016405E
Flash real size: 4194304 bytes

Flash Chip configuration ok.
Sketch size: 610048
Sketch free: 2535424
FreeHeap: 38792
Initializing EffectsManager
Initializing Settings
SPIFFS Settings file exists: true
SPIFFS Settings file size: 635
SPIFFS Effects file exists: true
SPIFFS Effects file size: 6271
Initializing MyMatrix
Set max brightness to: 255
Set current limit to: 1500
Set rotation to: 3
Framebuffer_GFX::begin Width: 16 Height: 16 Num Pixels: 256
Initializing web server at: 80
AutoConnect started

AP IP address: 
8.8.8.8
Starting DNS server: success
AutoConnect finished
Initializing LocalDNS
mDNS responder (firelamp) started!
Announcing http (tcp) service on port 80
Setup mode entered. No effects!
Scan done
Found networks: 2
```

I don't understand. How do I upload "spiffs". And where do I define the wifi settings?

Sorry, please be kind to a newbie :D

CODeRUS, 2020-05-25:
From logs it seems to be working now :)

CODeRUS, 2020-05-25:
notice now you have `SPIFFS Effects file exists: true` but in previous log it was `SPIFFS Effects file exists: false`

fitim7, 2020-05-25:
Ok, i understand this :)...
But now how do I connect this with my wifi? I can't find any setting for this. If I just connect to the hotspot called "Fire Lamp" nothing happens, just this in monitoring:
```
Request redirected to captive portal: /wifi.html
Request redirected to captive portal: /wifi.html
Request redirected to captive portal: /wifi.html
Request redirected to captive portal: /wifi.html
Request redirected to captive portal: /wifi.html
Request redirected to captive portal: /wifi.html
```

CODeRUS, 2020-05-25:
you have wifi.html in your data, right?

fitim7, 2020-05-25:
no, I just have effect.json and settings.json

CODeRUS, 2020-05-25:
please grab the latest data and extract all of it to your data folder :)

fitim7, 2020-05-25:
Yes, I got it :).. Thank you sir!

Can I ask you another thing. Is it possible to upload the .bin files directly with Platformio? If yes: how? :D

CODeRUS, 2020-05-25:
sure, go here:
[image]

fitim7, 2020-05-25:
Thank you mate. It is working perfect. I just have problems with the touch sensor. Mine is connected to GPIO4 (D2) on a NodeMCU. Are these the settings to change for it:

```
  "button": {
    "pin": 0,
    "type": 0,
    "state": 0
```
What do those variables mean? What do I insert for GPIO4 (D2)?

CODeRUS, 2020-05-25:
set pin: 4 and type: 1

type and state are described here: https://github.com/CODeRUS/GyverLampCpp/blob/master/lib/GyverButton/GyverButton.h#L47

fitim7, 2020-05-25:
> set pin: 4 and type: 1
> 
> type and state are described here: https://github.com/CODeRUS/GyverLampCpp/blob/master/lib/GyverButton/GyverButton.h#L47

type:2 worked for me! Thank you for your support :)

closed 2020-05-25 by fitim7