CODeRUS/GyverLampCpp #20: Build failure caused by utility library

BlipRanger, 2020-07-11:
Hello, I was attempting a quick build and ran into the following error:

`.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\algorithm:60:30: fatal error: 
utility: No such file or directory
 #include <utility> // UK-300.
                              ^
compilation terminated.
*** [.pio\build\nodemcu\lib635\FastLED\hsv2rgb.cpp.o] Error 1`

Is this a common problem or just user error on my part? 
My steps taken were: clone repo -> import project -> build

CODeRUS, 2020-07-11:
well, Travis says build is okay: https://travis-ci.org/github/CODeRUS/GyverLampCpp
try `pio update` and `pio platform update`

BlipRanger, 2020-07-11:
No luck with the updates, but I'll chalk it up to my own failings. The esp32 env worked fine, so I likely just have something set up incorrectly.

CODeRUS, 2020-07-11:
try removing that broken package folder and run build again

BlipRanger, 2020-07-11:
> 
> 
> try removing that broken package folder and run build again

That worked! Thank you!

closed 2020-07-11 by BlipRanger