CODeRUS/GyverLampCpp #21: Homeassistant MQTT setting effects doesn't change the effect on the device

BlipRanger, 2020-07-12:
I've encountered a strange issue where sending an effect via MQTT does not change the effect. The output from the devices is as follows:

```
homeassistant/light/firelamp/set
{
  "state": "ON",
  "effect": "Lava 3D"
}
Activating effect[17]: Lighters
Sending state
homeassistant/light/firelamp/state
{
  "state": "ON",
  "brightness": 126,
  "speed": 127,
  "scale": 10,
  "effect": "Lighters"
}
State sent: success
Sending state
homeassistant/light/firelamp/state
{
  "state": "ON",
  "brightness": 126,
  "speed": 127,
  "scale": 10,
  "effect": "Lighters"
}
State sent: success
Saving settings... Done!
Saving effects... Done!
```
If I change the active effect in the settings.json and push that over to the device, it does actually change the effect.

romanovroman74, 2020-07-12:
I have the same problem.

CODeRUS, 2020-07-12:
workin' on it

CODeRUS, 2020-07-12:
Fixed in bc4fc616231c3259e33ef6ee50a204874159a396

closed 2020-07-12 by CODeRUS