CODeRUS/GyverLampCpp #22: Effects are sometimes are rotated incorrectly

BlipRanger, 2020-07-13:
Every so often an effect like Fire or BouncingBalls will be rotated 90 degrees from the set rotation. Additionally, FactorialFire almost always seems to be 180 degrees rotated from the given bottom. Is there a method to send rotation via MQTT?

CODeRUS, 2020-07-13:
Hello!

It seems you incorrectly configured your matrix. If possible attach photo oor video when matrix test boot animation is played.
Test animation should be played from bottom left corner of matrix, red pixel should move horizontally to right, green pixel vertically to left.

BlipRanger, 2020-07-13:
Correct, that is how the boot animation looks for me. It's just a weird momentary glitch when changing from one effect to another, sometimes it rotates it. Once you switch again it goes away. I haven't been able to catch it on video yet though. Not a big deal, just wanted to bring it to your attention. I'll keep trying to recreate the issue.

CODeRUS, 2020-07-13:
clock effects have adjusted rotation, does you switch from clock effects?

BlipRanger, 2020-07-13:
No, it was transitioning from light balls to fire I think.

CODeRUS, 2020-07-19:
Fixed in 809d2f2c8f89774251d6d8d385b215734003d1e8

closed 2020-07-19 by CODeRUS

BlipRanger, 2020-07-23:
Thanks! You're the best!