CODeRUS/GyverLampCpp #26: Instruction for mqtt and button

nimda5, 2020-07-23:
First for the great fork, was using for long time.
Today updated to the latest: 20200719143431-809d2f2
MQTT command for power is working, but to changing effect - something changed...
`homeassistant/light/firelamp/set
{"effect":"Fire"}
`
Effects is not changing anymore.

Another question:
what to provide in config to make original Gyver D2 button (on mini)  working. At least effect change and power. 
`
  "button": {
    "pin": 0,
    "type": 0,
    "state": 0
  }
`

CODeRUS, 2020-07-23:
for mqtt you should send also `"state": "ON"` as it designed for Home Assistant usage, mqtt topics are optimized for it

D2 is GPIO4 on your board, so put pin: 4

nimda5, 2020-07-23:
I using OpenHab little bit different and more manual job ;)
Correct me if wrong:
`{"effect":"Fire"; "state":"ON"}'
and for button:
 "button": { "pin": 4, "type": 0, "state": 0 }

CODeRUS, 2020-07-23:
its json, so comma instead of semicolon

nimda5, 2020-07-23:
Yes, sorry - my fall.
Changed button to pin 4, and can't get working anymore. After restart lamp is switching on and then going off 
(I see wifi connection on router)
Opening from web interface - red header:
[image]

CODeRUS, 2020-07-23:
check uart logs

nimda5, 2020-07-23:
I'm not so smart :) 
Uart is serial?

CODeRUS, 2020-07-23:
connect usb to pc and run serial monitor :))

nimda5, 2020-07-23:
I have no idea how but I got it working. Powering ON was pressing button random times :) Magic.. Now working.
Checked and effect change working: `{"effect":"Fire", "state":"ON"}`
But interesting and brightness was working without state, but seams it's HO feature too.

P.S. just as simple idea: add Random to effect list as one of effects ;)
Thanks again - great job.
This ticket can be closed ;)

CODeRUS, 2020-07-23:
random can't be done via effect, can be implemented with external command. please post separate issue for this, i'll take a look later

closed 2020-07-23 by CODeRUS