CODeRUS/GyverLampCpp #28: Matrix flickering when wifi/webserver is used on ESP32

BatsIhor, 2020-08-08:
Hey, 

Used it for a bit and looks like due to some timing issues it flickers a lot on ESP32.
Researched that topic a bit and looks like it's common issue with ESP32 and one of the solutions is to use second core to run all the animations. 
[https://esp32.com/viewtopic.php?f=2&t=3980&sid=292952fc5eae3d5ec571838746448e39&start=10](url)

Here is possible solution:
[https://electronics.stackexchange.com/questions/375669/ws2812-esp32-flickering-math-issue](url)

If I manage to fix it I will create pull request. Let me know if you have any other solution to try. 

Thank you

CODeRUS, 2020-08-08:
for me flickers are gone if you close web ui

BatsIhor, 2020-08-13:
Yep that helps but, not a solution. Testing locally execution of animations from Core0.

CODeRUS, 2020-08-14:
please check dc4619192d39dbc03a8d0f113b6e50e3d16460b2

BatsIhor, 2020-08-14:
Thank you, Going to check it a bit later today.

CODeRUS, 2020-08-17:
@BatsIhor any sucess? :D

BatsIhor, 2020-08-18:
Yep, it works. Most of the flickering is removed. There are some cases when it still flickers but it's minor. 
Thank you

CODeRUS, 2020-08-18:
great, thanks for heads up!

BatsIhor, 2020-09-22:
So after suing it for quite a bit I see issue is not fixed. I also noticed that other popular platform WLED doesn't have that issue on esp32 plus they don't use multiple cores. So I think that might be related to matrix driver we use. I'm still looking into it but don't have enough experience to do good analysis. So I will keep this issue open and hopefully we will figure it out.

CODeRUS, 2021-01-04:
@BatsIhor seems fastled team fixed this problem in master. please check #42 branch

closed 2021-01-12 by CODeRUS