CODeRUS/GyverLampCpp #29: Losing MQTT Connection

fitim7, 2020-08-24:
I have connected the lamp via MQTT to Home Assistant. This is working for a few hours, then losing the connection. Everything else is working perfect. 
I always need to restart (turning power off and on) the lamp to connect again.

Why is this happening?

CODeRUS, 2020-08-24:
can you capture uart log of that period? also tell if web interface still available

fitim7, 2020-08-24:
How do I capture "uart log"? :D ... I think im a noob in this.
Yes, web interface is working perfect ...

CODeRUS, 2020-08-24:
i mean serial monitor mesages, via usb

fitim7, 2020-08-25:
Hmm..Unfortunately I'm not able to connect it without problems, because the lamp is permanently fixed already...

Is there another way to check the logs?

fitim7, 2020-08-25:
I use a NodeMCU, maybe this is relevant..

CODeRUS, 2020-08-25:
can you test previous version? i changed mqtt backend library in recent version, probably it's not stable enough

fitim7, 2020-08-25:
Hmm.. I'm using your firmware since 6 months maybe... already had this issue, also with previous versions. I just updated, because I thought the new mqtt libraries will fix this :D ... which unfortunately didn't :D

CODeRUS, 2020-08-25:
Weird then. Is there possibility of the problem in your mqtt broker?

fitim7, 2020-08-25:
To be honest: I don't think so. I got a bunch of mqtt devices, and all are working perfect...
Is there a possibility to implement logs via the web interface?

top4ek, 2020-08-25:
Heh. I'm not alone. 
I've blamed my old keenetic giga which drops clients from time to time and forbids them to reconnect for couple of minutes. I have firmware somwhere from spring too and, as i see in HA logs, connection lives for 3-4 hours. Also i don't see problems with mqtt — more 20 zigbee devices working through it.

Now I'm running monitor. Waiting for results.

CODeRUS, 2020-08-25:
@top4ek you mean you have just one (z2m) client connected to broker locally, right?)

top4ek, 2020-08-25:
Got:
```
MQTT disconnect reason: TCP_DISCONNECTED
Connecting ot MQTT...
```
And nothing in 10 minutes. Web interface works fine. 
 
@CODeRUS Oh, yes :) only one z2m. But i have two lamps with our firmware and one sonoff basic with ESPHome... May be I'll try ESPHome via MQTT today(it uses it's own protocol by default).

CODeRUS, 2020-08-25:
I can try to reproduce at weekend

top4ek, 2020-08-25:
Why I didn't try that before? :) When i disable network or turn off router I see same behaviour.

CODeRUS, 2020-08-25:
So after reconnecting to wifi mqtt connection didn't recover?

top4ek, 2020-08-25:
Yep.

fitim7, 2020-08-25:
It sounds silly: But I'm glad that someone else has this problem! :D:D

CODeRUS, 2020-08-26:
Added improvements to latest master, please test

fitim7, 2020-08-26:
I will test immediately... thx for your effort 😊

fitim7, 2020-08-26:
After the update it doesn't connect to mqtt anymore ... also not after restart

CODeRUS, 2020-08-26:
you compiling from sources? try removing .pio cache folder and compiling from scratch

top4ek, 2020-08-27:
Neither mine.
```
Local ip: 192.168.0.5
AutoConnect finished
Initializing LocalDNS
mDNS responder (bedlamp) started!
Announcing http (tcp) service on port 80
Initializing TimeClient
Initializing GyverTimer: 192.168.1.3, offset: 10800
Initializing MqttClient
Activating effect[0]: Fire
Connecting to MQTT...
Scan done
Found networks: 4
Saving settings... Done!
Saving effects... Done!
MQTT disconnect reason: unknown -8

```
Web still works(awesome redesign btw)
Unable to google that reason in sources. O_o
Why it scans for networks after  successfuly connected?

When turning down and up it's wifi network:
```
Disconnected from Wi-Fi.
Disconnected from Wi-Fi.
Disconnected from Wi-Fi.
Connected to Wi-Fi.
Connecting to MQTT...
MQTT disconnect reason: unknown -8
```

CODeRUS, 2020-08-27:
add `#define PANGO_DEBUG 1` into MqttCLient.cpp before `#include <PangolinMQTT.h>` and rebuild

CODeRUS, 2020-08-27:
> Besides, getting occasional segfaults when switching effects. Maybe it's time to move to ESP32? :)

same on esp32, some efffects are not polished. if some effects change crash 100% reproduceable, i can fix

top4ek, 2020-08-27:
Strange shit happening :) 
After adding DEBUG option, lamp successfully connected. Removing it back doesn't break anything. I dunno, will continue monitoring. I don't beleive, that housing disassemble(found a "bug" too) or moving it over table have repaired mqtt  :)

BTW, after that i've compiled it from scratch: works fine for a half of hour already.  ¯\\_(ツ)_/¯

BUT, I've noticed that if lamp is unable to connect it fallbacks to AP mode and after that it don't try to reconnect back even with power cycle. Is there any way to track presence of configured network while AP mode and connect to it if found?

```
some efffects are not polished
```
Prismata, i beleive, polished. It was running last 3 monthes almost every day :)

```
same on esp32
```
Eeh :) Have ordered three units today.

CODeRUS, 2020-09-12:
wifi credentials are erased if access point was not found. please open separate issue for adjusting wifi credentials workflow

BlipRanger, 2020-10-12:
I saw something similar today on the HA side. Simply a MQTT message "1602511936: Socket error on client FireLampClient-devlamp8266, disconnecting."
The device reconnected after power cycling it. Any chance this is related? Does the client attempt a reconnect after a period of time?

fitim7, 2020-10-13:
Since the update it is working for me! Also on nodemcu

fitim7, 2020-10-15:
Broke again...

CODeRUS, 2020-12-22:
Please check with latest commits

fitim7, 2020-12-22:
I'm not able to connect atm. I don't why. If I enter my wifi credentials in settings, it still doesn't connect but just starts with AP Mode. How can I turn this off? Somehow I am not able to enter with AP Mode...

CODeRUS, 2020-12-22:
set correct button config ot set 255 to pin number of button. holding button on boot erasing wifi credentials and starting AP mode

fitim7, 2020-12-23:
The problem is, that somehow It is not flashing the whole settings... for example it is also not changing rotation. I erased the flash, clean build the firmware.bin and fs.bin and uploaded. it is always holding the "old" settings..

CODeRUS, 2020-12-23:
you uploading flash via cable using uploadfs?

fitim7, 2020-12-23:
Yes. This is my settings.json:

`
{
  "activeEffect": 0,
  "logInterval": 0,
  "matrix": {
    "width": 16,
    "height": 16,
    "segments": 1,
    "type": 15,
    "maxBrightness": 255,
    "currentLimit": 1500,
    "rotation": 1
  },
  "connection": {
    "mdns": "firelamp",
    "apName": "Fire Lamp",
    "ntpServer": "europe.pool.ntp.org",
    "ntpOffset": 10800,
    "hostname": "firelamp",
    "ssid": "qerrtqert",
    "password": "qertqert"
  },
  "mqtt": {
    "host": "192.168.178.40",
    "port": 1883,
    "username": "eqrtqert",
    "password": "qerrterqrt",
    "uniqueId": "firelamp",
    "name": "firelamp",
    "model": "firelamp"
  },
  "button": {
    "pin": 4,
    "type": 1,
    "state": 0
  }
}

`

And monitoring after "pio run --target uploadfs -e nodemcu" says this:

Happy debugging!
Flash ide  size: 4194304 bytes
Flash ide speed: 40000000 Hz
Flash ide mode:  DIO
Flash real id:   0016405E
Flash real size: 4194304 bytes

Flash Chip configuration ok.
Sketch size: 618256
Sketch free: 2527232
1_FreeHeap: 35856
Initializing Settings
FLASHFS Settings file exists: true
FLASHFS Settings file size: 510
Button pin: 4
Initializing EffectsManager
FLASHFS Effects file exists: true
FLASHFS Effects file size: 3921
Initializing MyMatrix
Set max brightness to: 255
Set current limit to: 1500
Set rotation to: 3
Framebuffer_GFX::begin Width: 16 Height: 16 Num Pixels: 256
Initializing web server at: 80
AutoConnect started

Starting AP: Fire Lamp
AP started
AP IP address: 8.8.8.8
Starting DNS server: success
scheduleScan
AutoConnect finished
Activating effect[0]: Fire
Scan done
Found networks: 8
Saving settings... Done!
Saving effects... Done!
FrameBuffer::GFX 11fps
FrameBuffer::GFX 27fps
FrameBuffer::GFX 27fps

CODeRUS, 2020-12-23:
okay, then try to hold button while booting, you should see `!!! Setup mode entered. No effects !!!` message, then reboot normally

fitim7, 2020-12-23:
Just tried, still the same:

!!! Setup mode entered. No effects !!!
Initializing web server at: 80
AutoConnect started

Starting AP: Fire Lamp
AP started
AP IP address: 8.8.8.8
Starting DNS server: success
scheduleScan
AutoConnect finished
Scan done
Found networks: 5
{$␀lܞ|␀�c␃�N␌␌␄�␌l�␄b|��␂�p␃��|␒c�␄"�p␡�N�loN���␌#␜p�cl ␂rlp�N�␐␃␄␄�␌d␄��␄␄␌c␄N��o�$�␌�p␄��oo�␂l��$`␃�␒␛oo␄l`␂␎␂n{����␃␌␌b␃l`␂␎s��N␄␌B␂$`␃�b␌␌��$�lr� ␃��o�␂
Happy debugging!
Flash ide  size: 4194304 bytes
Flash ide speed: 40000000 Hz
Flash ide mode:  DIO
Flash real id:   0016405E
Flash real size: 4194304 bytes

Flash Chip configuration ok.
Sketch size: 618256
Sketch free: 2527232
1_FreeHeap: 35856
Initializing Settings
FLASHFS Settings file exists: true
FLASHFS Settings file size: 511
Button pin: 4
Initializing EffectsManager
FLASHFS Effects file exists: true
FLASHFS Effects file size: 3922
Initializing MyMatrix
Set max brightness to: 255
Set current limit to: 1500
Set rotation to: 3
Framebuffer_GFX::begin Width: 16 Height: 16 Num Pixels: 256
Initializing web server at: 80
AutoConnect started

Starting AP: Fire Lamp
AP started
AP IP address: 8.8.8.8
Starting DNS server: success
scheduleScan
AutoConnect finished
Activating effect[0]: Fire
Scan done
Found networks: 7
Saving settings... Done!
Saving effects... Done!
FrameBuffer::GFX 11fps

CODeRUS, 2020-12-23:
can you try editing file `.pio\libdeps\nodemcu\ESPReactWifiManager\ESPReactWifiManager.cpp`
line 324, first line of `ESPReactWifiManager::connect` function
replace `Serial.println();` with `Serial.println(ssid);`
build and upload, check log

fitim7, 2020-12-23:
Just tried, nothing changed. I thing it has something to do with the settings, maybe it doesn't load them .. because if I change rotation for example to 1 or 2 , it still stays at 3. I don't know what to to anymore ... :D

CODeRUS, 2020-12-23:
it should show ssid after `AutoConnect started` instead of empty line.
did you tried removing everything or clonning/downloading to new folder? :)
also please show log of uploadfs. make sure no other esp board is connected :))

fitim7, 2020-12-23:
[image]
Just to have no misunderstanding: In Platform IO there is the "General" Build and Upload and Clean, and the "Platform" Build FS, Upload FS and Erase Flash. Do I need to build both and upload both, or just the Filesystem?

CODeRUS, 2020-12-23:
upload both. for clean its better jsut to remove .pio folder manually

CODeRUS, 2020-12-23:
just in case try to upload filesystem using commandline

fitim7, 2020-12-23:
Is the order important? First General, then FS? Or first FS then general?

CODeRUS, 2020-12-23:
firmware then filesystem

CODeRUS, 2020-12-23:
ping me tomorrow, will try to debug

fitim7, 2020-12-23:
My friend ... you will laugh at me :D ... I was editing the settings.json in project folder and not the data/settings.json ... everything is working now. This settings.json is confusing in the project folder, since it looks exactly the same :D ...

Thank you for your time and help!

CODeRUS, 2020-12-23:
There is slould be no settings.json outside data folder in my project. But anyway, congrats)) let me know about mqtt issues

fitim7, 2020-12-23:
> There is slould be no settings.json outside data folder in my project. But anyway, congrats)) let me know about mqtt issues

Lol, this sentence showed my whole mistake... I unzipped data.zp from release page to the project folder instead of the data folder... my mistake. Just did a clean install. Will test mqtt, but working for maybe 15 hours without problems now.

CODeRUS, 2021-01-12:
Updated mqtt library, added force reconnect on failures, seems stable for me.

closed 2021-01-12 by CODeRUS