CODeRUS/GyverLampCpp #30: data

edgiru, 2020-08-28:
Подскажите сами архивы *.gz надо распаковывать в папку data?
Или копировать в таком виде в котором есть?
[image]

CODeRUS, 2020-08-28:
Нужно оставить как есть

edgiru, 2020-08-28:
> Нужно оставить как есть

Тогда не понятно почему страницы не грузятся. web сервер стартует, а не чего не грузиться.
логи консоли:
`
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on COM3  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
load:0x3fff0018,len:4

Happy debugging!
Flash ide  size: 4194304 bytes
Flash ide speed: 40000000 Hz
Flash ide mode:  DIO
Sketch size: 1131776
Sketch free: 1310720
Total heap: 337764
Min free heap: 304364
Max alloc heap: 113792
FreeHeap: 311244
Initializing Settings
FLASHFS Settings file exists: true
FLASHFS Settings file size: 484
Button pin: 0
Initializing EffectsManager
FLASHFS Effects file exists: true
FLASHFS Effects file size: 3921
Initializing MyMatrix
Set max brightness to: 255
Set current limit to: 1500
Set rotation to: 3
Framebuffer_GFX::begin Width: 16 Height: 16 Num Pixels: 256
Initializing web server at: 80
AutoConnect started

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

Happy debugging!
Flash ide  size: 4194304 bytes
Flash ide speed: 40000000 Hz
Flash ide mode:  DIO
Sketch size: 1131776
Sketch free: 1310720
Total heap: 337764
Min free heap: 304848
Max alloc heap: 113792
FreeHeap: 311244
Initializing Settings
FLASHFS Settings file exists: true
FLASHFS Settings file size: 484
Button pin: 0
Initializing EffectsManager
FLASHFS Effects file exists: true
FLASHFS Effects file size: 3921
Initializing MyMatrix
Set max brightness to: 255
Set current limit to: 1500
Set rotation to: 3
Framebuffer_GFX::begin Width: 16 Height: 16 Num Pixels: 256
Initializing web server at: 80
AutoConnect started

[WiFi-event] event: SYSTEM_EVENT_WIFI_READY
[WiFi-event] event: SYSTEM_EVENT_STA_START
Connecting to last saved network
Connecting to network: oc
[WiFi-event] event: SYSTEM_EVENT_STA_CONNECTED
[WiFi-event] event: SYSTEM_EVENT_STA_GOT_IP
Connected to Wi-Fi.
STA IP address: 192.168.1.90
scheduleScan
AutoConnect finished
Initializing LocalDNS
mDNS responder (firelamp) started!
Announcing http (tcp) service on port 80
Initializing TimeClient
Initializing GyverTimer: europe.pool.ntp.org, offset: 10800
Initializing MqttClient
Empty host for MqttClient
Activating effect[0]: Fire
[WiFi-event] event: SYSTEM_EVENT_SCAN_DONE
Scan done
Found networks: 10
Saving settings... Done!
Saving effects... Done!

top4ek, 2020-08-28:
Бывает по какой то причине 8266 не хочет быстро отдавать ассеты. Подрочив браузер, ассеты таки отдаются и в следующий раз не качаются. Такое ощущение, что просто не хватает мощности. Но на последних версиях я не замечал проблем.

CODeRUS, 2020-08-28:
Можно включить вывод кучи в логе, скорее всего памяти мало. Можно отключить ненужные эффекты, чтобы было полегче

edgiru, 2020-08-28:
это esp32

CODeRUS, 2020-08-28:
На 32 вообще никаких проблем нет у меня. Попробуй очистить и начисто все прошить снова

edgiru, 2020-08-28:
Спасибо за ответы. Несколько раз чистил сейчас опять еще раз почистил и прошил все заработало, на компе не открывается страница только не телефоне. @CODeRUS  Вы похоже из Татарстана. Я из Казани ))

closed 2020-08-28 by edgiru

CODeRUS, 2020-08-28:
по айпи вычислил?) я в Иннополисе живу.

edgiru, 2020-08-28:
на скриншоте web интерфейса Тателеком точки доступа. )))