CODeRUS/GyverLampCpp #31: After updating to PlatformIO 5 it fails to build

BatsIhor, 2020-09-11:
Hi there, 

I just noticed that after update to PlatformIO 5 it fails to build project. 
.pio/libdeps/nodemcu/FastLED NeoMatrix/FastLED_NeoMatrix.cpp:37:26: fatal error: Adafruit_GFX.h: No such file or directory

I'm not sure where the problem is, but might be actually issue they introduced and not something you can fix, so will ask on their forum as well. 

Thank you 
Ihor

top4ek, 2020-09-11:
Tried to compile — the same. May be it's related to "GrayLED" update in this lib, didn't bury enough.

CODeRUS, 2020-09-11:
interesting, i'll check, thanks

CODeRUS, 2020-09-12:
Fixed with https://github.com/CODeRUS/GyverLampCpp/commit/8ec97e392cc66ac8739199163e6236223b4d167e

closed 2020-09-12 by CODeRUS