CODeRUS/GyverLampCpp #33: ESP8622: Exception (28) when use button for navigating through the effects.

ago-cs, 2020-09-22:
The problem occurs during switching effects via button double- or triple-clicks (in the React web app all works fine).

**Steps to reproduce**
1. Plug lamp to DC
2. Plug lamp to USB
3. Switch the lamp ON
4. Start navigating using button double- or triple-clicks.
5. after a while -> Exception (28): LoadProhibited
6. Reboot

Not 100% but very-very often going up through the sequence _Snow_ > _Matrix_ **>(EXCEPTION)>** _Lighters_: Exception thrown after the attempt to switch from Matrix to Lighters the exception (decoded via ESP Exception Decoder tool): 
```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
PC: 0x40214e7f
EXCVADDR: 0x00000000
```
Tried to get more info with -Og flags but it didn't help - this is the only I could get.

**Hardware:**
- NodeMCU v2
- Arduino button module (3 contacts): signal wire D2 / GPIO4 (type 1)
- Led Matrix for GyverLamp: D4 / GPIO2 (default settins)
- DC 5V / chineese 3A linked to Vin, USB cable.

**Settings:**
```
{
  ...
  "matrix": {
    "width": 16,
    "height": 16,
    "segments": 1,
    "type": 15,
    "maxBrightness": 255,
    "currentLimit": 1500,
    "rotation": 3
  },
  "button": {
    "pin": 4,
    "type": 1,
    "state": 0
  }
}
```

mykey-tech, 2020-09-25:
...same issue but with random effects

ago-cs, 2020-09-25:
Some additional details:
* Time period between switches doesn't make sense.
* Full clean of module memory by before flashing 
* @mykey-tech , you right: the failure happens during switching of effects randomly, but the Matrix is 99% reproducible. 

As an idea of a dirty fix, I will try to go the same way as switching from web/MQTT requests.
Will be glad to any other ideas.

Alexandr180, 2020-11-04:
Аналогичная проблема, перезагрузка контроллера при переключении некоторых эффектов, решил временно убрав функцию "deactivate effect" стало норм, но наверное может приводить к утечке памяти...

CODeRUS, 2020-11-04:
thanks! should be fixed now

commit 3614bbe0 (2020-11-04): [effectsManager] Fix switching effects with button. Fixes #33
closed 2020-11-04 by CODeRUS