CODeRUS/GyverLampCpp #34: Touch sensor not working with nodemcu

mykey-tech, 2020-09-24:
build your project few minutes ago.
everything works fine except the touch sensor 
can you tell me why?

CODeRUS, 2020-09-24:
You didn't fill button config on settings.json?

mykey-tech, 2020-09-24:
.thnx for the quick reply but.....
what do i have to setup  for
pin, type, state

sincerely mike

mykey-tech, 2020-09-24:
lash Chip configuration ok.
Sketch size: 617808
Sketch free: 2527232
FreeHeap: 37376
Initializing Settings
FLASHFS Settings file exists: true
FLASHFS Settings file size: 484
Button pin: 0
Initializing EffectsManager
FLASHFS Effects file exists: true
FLASHFS Effects file size: 3921
Initializing MyMatrix
Set max brightness to: 255
Set current limit to: 1500
Set rotation to: 3
Framebuffer_GFX::begin Width: 16 Height: 16 Num Pixels: 256
Initializing web server at: 80
AutoConnect started

CODeRUS, 2020-09-24:
thanks for head up, added configuration explanation to readme!

closed 2020-09-24 by CODeRUS

mykey-tech, 2020-09-24:
...but button pin still low
doesn't work

CODeRUS, 2020-09-24:
what is your config for button? what kind of button is used? ttp223? touch is high or low?

mykey-tech, 2020-09-24:
ok it is the ttp223
touch is high
"button": {
    "pin": 4,
    "type": 1,
    "state": 0
  }

CODeRUS, 2020-09-24:
make sure you define GPIO pin number, for example:

D4 is GPIO 2 for nodemcu

[image]

mykey-tech, 2020-09-24:
yes i did ..used the same pinout picture as you
but nothing happened is it right that in the serial console
Button pin: 0
?

CODeRUS, 2020-09-24:
seems you didnt uploaded filesystem image?

CODeRUS, 2020-09-24:
you can edit settings.json and upload it via web ui as well

mykey-tech, 2020-09-24:
yeah that was the problem
i'm dump sorry
sincerely mike