CODeRUS/GyverLampCpp #35: Single color effect color format

nimda5, 2020-10-10:
Please little help: when posting color for single effect - 7 digits is send.
What is the name of this color format? I need to convert if from OH to make work.

CODeRUS, 2020-10-10:
Decimal format

nimda5, 2020-10-10:
Ok, now got idea.
#3ab429 is 58,180,41 and in mqtt: 3847209
Left only how to convert OH color item to decimal.

CODeRUS, 2020-10-10:
(r << 16) + (g << 8) + b

But home assistant mqtt should work with rgb

nimda5, 2020-10-10:
Big thanks. 
I'm on OH, there is another trick to get rbg and then will convert to decimal.
One more time thanks - great job.

closed 2020-10-10 by nimda5

CODeRUS, 2020-10-10:
But you can still use mqtt topics for home assistant integration ;)

nimda5, 2020-10-10:
Yes, sure. I plan to move to the HA ;) OH is not goint a way I think they have to go.

nimda5, 2020-10-11:
Maybe somebody will need rule for OH to convert color to digit:
```
rule "Modify color to rgb and then digital"
  when
    Item Fire_Lamp_Single_Color received command
  then
	val red = ((receivedCommand as HSBType).red * 2.55).intValue
	val green = ((receivedCommand as HSBType).green * 2.55).intValue
	val blue = ((receivedCommand as HSBType).blue * 2.55).intValue
	var rgb = red
	rgb = (rgb << 8) + green
	rgb = (rgb << 8) + blue
	Fire_Lamp_Single_Color_Send.sendCommand("{\"state\":\"ON\", \"effect\":\"Single color\", \"color\":" + rgb.toString(10) + "}")
end