CODeRUS/GyverLampCpp #36: Проблемы с разрешением матрицы отличным от 16х16.

Alexandr180, 2020-11-04:
Спасибо за Вашу работу! Отличный проект! Искал проект для platfirmio, и был очень приятно удивлен!
Использую ленту ws2812b (аналог) намотанную на трубе, итоговое разрешение 6х45.
Привел параметры в соответствие, но Fire2012 и Fie2018 вызывают перезагрузку. Просто "Fire effect" работает отлично после правки hueMask и valueMask, и некоторых значений в параметрах эффекта.
С матрицей 16х16 этой проблемы нет.
Подскажите где копать?

CODeRUS, 2020-11-04:
Очень интересно, спасибо, смотрю

CODeRUS, 2020-11-04:
Нашел очень интересную ошибку в коде, из разряда: как оно вообще работало?))

closed 2020-11-04 by CODeRUS

Alexandr180, 2020-11-11:
Удивительно но Fire2012 заработал с моим разрешением ленты-матрицы а Fire2018 как и раньше приводит к перезагрузке. Похоже что разрешение больше чем 16х16 поддерживается а вот 45х6 нет. и 16х6 тоже нет. Просмотрел код, чувствую себя ущербным и не могу разобраться )))) Буду очень благодарен если подскажите где копать!

CODeRUS, 2020-11-11:
Починил Огонь 2018, спасибо!

commit a43a0d4a (2020-11-04): [matrix] Fix for wrapX, wrapY. Fixes #36