CODeRUS/GyverLampCpp #37: Просьба добавить 2е функции: 1. Сброс параметров эффектов и 2. Сброс общих настроек (в том числе wifi).

Alexandr180, 2020-11-07:
Пожалуйста сделайте по возможности 2 функции сброса по кнопке, (или подскажите "на пальцах" как это лучше сделать?). Например -
1. Удержание кнопки при включении питания приводит к полному восстановлению spiffs.
2. Удержание кнопки на 10 секунд в обычном режиме приводит к сбросу параметров эффектов.

Зачем это нужно?
1. Иногда раз в несколько дней при интенсивном использовании, и перещелкивании эффектов, происходят какие то ошибки записи в spiffs, и лампа уходит в циклическую перезагрузку, веб интерфейс не доступен и восстановление возможно только по шнурку, перезаливкой spiffs.
2. Для некоторых конечных пользователей сложно после попыток порегулировать разные параметры эффектов вернуть все к приятному первоначальному виду.
p.s. Очень нравится работа заливки прошивки/spiffs/settings  по воздуху. делал это десятки раз и ни разу "не положил" лампу! Да иногда не с первого раза заливается, но в общем результат всегда достижим!

top4ek, 2020-11-07:
На ESP32 это реализуемо, там флешки совковыми лопатами не выгрести, а вот на 8266 у которой 1 метр флешки уже может не получится(не у всех четырёхметровые, особенно в какой нить Sonoff basiс). 

Вообще странно, от порчи FS должен был помочь переход на LittleFS, но один раз с битым конфигом тоже столкнулся, причём на ровном месте, настройки я обычно не тыркаю, выключил ночью, а утром уже загрузиться не смогла. Видно по "инициализации" матрицы, что конфиг не подсосался :(

Как вариант, можно рядом с settings.json хранить что то эталонное, он много не весит и, в случае если оригинал мы прочитать не можем, то восстанавливать из этой копии. А эффекты уже 6 кило весят. Нать проверять :)

CODeRUS, 2020-11-07:
а смысл в чем? чтобы без компа/телефона сбрасывать? могу сделать как раз при удержании при включении чтобы запускался минимальный режим для заливки.

top4ek, 2020-11-07:
А как заливать? Если мы конфиг потеряли. Подключаться к дефолтной точке и там аплоадить?

CODeRUS, 2020-11-07:
Ну да к точке доступа

CODeRUS, 2020-11-16:
Добавил в https://github.com/CODeRUS/GyverLampCpp/commit/36b9965fbcf1877420e8618014708655c12d71f2
setupMode прерывает эффекты и запускает облегченный загрузчик

Обратите внимание на пины!
Кнопка подключенная к GPIO0/2/15, а это D3 D4 D8 на nodemcu будучи зажатой при загрузке не даст плате загрузиться.

[image]

Alexandr180, 2020-11-22:
С последним обновлением у меня при ошибке ввода пароля к точке доступа, esp32 не переходит в AP.
А хотелось бы.
AutoConnect started

[WiFi-event] event: SYSTEM_EVENT_WIFI_READY
[WiFi-event] event: SYSTEM_EVENT_STA_START
Connecting to last saved network
Connecting to network: Mywifi
[WiFi-event] event: SYSTEM_EVENT_STA_DISCONNECTED
[WiFi-event] event: SYSTEM_EVENT_STA_DISCONNECTED

И больше в логах ничего не появляется. висит пока не перезагрузишь.

CODeRUS, 2020-11-22:
А до этого переходила? Я добавлял только safemode по кнопке

Alexandr180, 2020-11-22:
Я и не пробовал неправильный пароль вводить )))
В библиотеке EspReactWifiManager.cpp (Библиотеку нашел у Вас же, понял что там тоже изменения)
в функции void checkRetryCount() добавил в условия IF (...|| WiFi.status() == WL_DISCONNECTED )
Теперь после нескольких попыток подключиться появляется в логах ошибка:
"Error starting AP". Может у меня с кодом что-то не так, пока не пойму.

CODeRUS, 2020-11-22:
В общем я вроде этот кейс ещё не смотрел. Пока в todo

Alexandr180, 2020-11-22:
Понял что из STA в AP esp32 не переходит без перезагрузки.
Если доступен SSID из сохраненных, то пытается подключиться к нему, а если не получилось перезагружается и все повторяется по новой. Что ни то придумаю - отпишусь.

CODeRUS, 2020-11-22:
Там есть AP+STA режим. Я пока а отпуске, через неделю сделаю.

CODeRUS, 2020-12-08:
Spiffs восстанавливать неоткуда
Зажатая кнопка при включении сбрасывает настройки wifi, включает точку доступа с простой страницей загрузки
Исправлены ошибки переподключения при пропаже точки

closed 2020-12-08 by CODeRUS