CODeRUS/GyverLampCpp #38: Более умная работа с WiFi

top4ek, 2020-11-07:
Сейчас при потере конфигурённой точки доступа прошивка дропает эти настройки и переходит в режим точки доступа. После этого приходится конфигурять вафлю заново. 

Чтобы пофиксить это нужно не ресетить данные, а просто переходить в режим точки доступа(а в идеале ещё и периодически проверять конфигурённую сеть на доступность). После перезагрузки прошивка опять должна долбится в конфигурённую сеть.

CODeRUS, 2020-11-07:
у меня в wip висит код для этого, надо дотестить, все руки не доходили)

top4ek, 2020-11-07:
Дык вот и у меня особо времени нет(этот тикет то собирался чуть ли не полгода написать). Да и руки у меня из жопы, когда доходит дело до C++ :)

CODeRUS, 2020-11-29:
Проверяй. Обновил модуль ESPReactWifiManager, можно просто удалить из папки .pio/libdeps, при сборке скачается новый

И еще добавил чтобы при включении с нажатой кнопкой очищались настройки wifi

closed 2020-11-29 by CODeRUS

top4ek, 2020-11-29:
Да ёлки :) 
Утром пару часов на эксперименты убил, вообще непонятная хрень творилась, животворящий erase только на первый раз помогал, а потом такое поведение, что fs тупо не прошивалась новая, хотел уже в телегу портянку накатать :) Но в итоге плюнул(т.к. подозрения на роутер), ~буду проституткой~ пошёл пить вискарь и прилетат это уведомление :)
В целом, всё огонь и присмата :) Пока полноценно не проверил, т.к. интернет домашним нужон, но учитывая непонятную херню с роутером, который не сразу подключает клиента, вроде как всё работает. Будем посмотреть, спасиба :)
Там ещё маааненький PR для home assistant, тоже слегка потрахал мозг, пока сообразил :)

CODeRUS, 2020-11-29:
Я с мобилы раздавал вайфай и отключал для теста :)

CODeRUS, 2020-12-07:
Еще закоммитил фиксов для esp32

top4ek, 2020-12-07:
У нас тут другая проблема вылазит: ломается файл с конфигом. Причём раньше я думал, что это из-за внезапного отключения во время записи(разве littlefs это не чинит?), а тут выяснилось, что оно и без обестачивания ломается.

CODeRUS, 2020-12-07:
> У меня пока нет ESP32 :) Взять взял, а пересобрать всё никак.
> У нас тут другая проблема вылазит: ломается файл с конфигом. Причём раньше я думал, что это из-за внезапного отключения во время записи(разве littlefs это не чинит?), а тут выяснилось, что оно и без обестачивания ломается.

Заведи баг на это опиши что к чему

top4ek, 2020-12-07:
100% воспроизвести не получается. Если вдруг, то да.

commit 7d2b6542 (2020-11-29): [wifi] Store correct fallback AP settings.
Requires updated ESPReactWifiManager module.
Fixes #38