CODeRUS/GyverLampCpp #43: Flickering on "Color" effect on low brightness

maksimkurb, 2021-01-12:
When I turn on *Color* effect on a low brightness *2-15* on a speed *1* lamp flickering very fast.
When I increases speed from *1* to *255* flickering disappears (at least from the eye perspective).

I wonder if it can be solved just via hardcoding *Color* effect speed to 255 or solution is more complicated?

I'm using Wemos D1 board

CODeRUS, 2021-01-12:
Thanks, fixed in 132f4b74acc4ee4147b5ae9101a32029b518d07a

closed 2021-01-12 by CODeRUS

maksimkurb, 2021-01-12:
Flickering is not solved because I can set speed to low values

CODeRUS, 2021-01-13:
There is nothing to do more. Leds are not capable for such mode, and FastLED library is doing dithering: https://github.com/FastLED/FastLED/wiki/FastLED-Temporal-Dithering

CODeRUS, 2021-01-14:
Added option to enable or disable dithering: 8532e40fca3212ac52f9b8b1e58f2563cecfb800
you can disable it if you want to