CODeRUS/GyverLampCpp #46: Settings and effects files get corrupted if ESP crash while writing

maksimkurb, 2021-01-12:
I'm testing this firmware on my lamp and sometimes my lamp restarting by watchdog or if it crashed (my matrix can not display clock).
If `settings.json` or `effects.json` were saving at this time, they get corrupted and I supposed to re-upload my settings.json and effects.json again to fix matrix/effects/mqtt/etc settings.
I think that these files should be written on the flash in a more safe way.

What came to my mind:
1. We upload `settings.json` and `effects.json` and ESP will use them in readonly mode. All updates will be saved to `settings.json.save` and `effects.json.save`. When lamp is booting up, it tries to read `*.json.save` files, and if they get corrupted, we can restore them from corresponding `*.json` originals. In this scenario, ESP should also delete `*.json.save` file if new `*.json` uploaded via web ui.
2. Make backups of running config. Now config is saving with 5 sec interval. We can also save `.bak` file every 60 seconds (if changed) for example, chance that both running-config and backup-config will be corrupted are lower
3. Play with `ESP.wdtEnable(timeout_ms)`? For now, timeout is 0ms, but I can't understand why, can you please explain it?
4. May be you have another ideas?

CODeRUS, 2021-01-12:
for 3: check https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Esp.cpp#L94
 software watchdog timers are not implemented at all in core, thats why we need to feed the dog sometimes.

CODeRUS, 2021-01-12:
For backups this is a good idea worth implementing

maksimkurb, 2021-01-13:
Another idea: postpone reboot (which is called via GET /reboot) if file is saving

CODeRUS, 2021-01-13:
> 
> 
> Another idea: postpone reboot (which is called via GET /reboot) if file is saving

i dont think /reboot is called by any user so precisely :))
but anyway sinbce i'll start implementing this i for sure guard all writings to flash

CODeRUS, 2021-01-14:
please check build from https://github.com/CODeRUS/GyverLampCpp/pull/49

CODeRUS, 2021-01-16:
Should be implemented. Please reopen if issues.

closed 2021-02-04 by CODeRUS