CODeRUS/GyverLampCpp #47: ESP8266 crashing when changing effect via MQTT to Rain Neo

maksimkurb, 2021-01-13:
Captured weird bug, when I change lamp effect 3 times to Rain Neo via MQTT (with interval about 1-2 sec), ESP crash.

I've captured crash logs, and decoded them via [ESP Exception Decoder](https://arduino-esp8266.readthedocs.io/en/latest/faq/a02-my-esp-crashes.html#exception).
I've got multiple different stacktraces, but first stacktrace was most frequent.

<details>
<summary>First stacktrace (frequent one)</summary>

```
Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
PC: 0x40100c4d
EXCVADDR: 0x40009b38

Decoding stack results
0x40246d55: operator delete(void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_op.cc line 48
0x40246d40: operator delete[](void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_opv.cc line 33
0x40246d40: operator delete[](void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_opv.cc line 33
0x4024bfcc: tcp_write at core/tcp_out.c line 666
0x4024719c: std::string::_Rep::_S_create(unsigned int, unsigned int, std::allocator  const&) at /workdir/arena.x86_64/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 621
0x4024719c: std::string::_Rep::_S_create(unsigned int, unsigned int, std::allocator  const&) at /workdir/arena.x86_64/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 621
0x40247264: std::string::_M_mutate(unsigned int, unsigned int, unsigned int) at /workdir/arena.x86_64/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 299
0x4024745c: std::basic_string  , std::allocator  >::basic_string(std::string const&) at /workdir/arena.x86_64/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 176
0x4024745c: std::basic_string  , std::allocator  >::basic_string(std::string const&) at /workdir/arena.x86_64/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 176
0x4024e364: etharp_output_to_arp_index at core/ipv4/etharp.c line 770
0x4024ab28: tcp_input at core/tcp_in.c line 501
0x4024ee75: ip4_input at core/ipv4/ip4.c line 1467
0x40247efd: ethernet_input_LWIP2 at netif/ethernet.c line 188
0x40247d23: esp2glue_ethernet_input at glue-lwip/lwip-git.c line 469
0x40279c2e: ethernet_input at glue-esp/lwip-esp.c line 365
0x40279c3f: ethernet_input at glue-esp/lwip-esp.c line 373
0x4024f708: mem_malloc at core/mem.c line 210
0x4024af06: pbuf_alloc_LWIP2 at core/pbuf.c line 284
```

</details>

<details>
<summary>First stacktrace (original Serial output)</summary>

```
Activating effect[54]: Rain Neo
Sending state
homeassistant/light/firelamp-max2/state
{
  "state": "ON",
  "brightness": 80,
  "speed": 70,
  "scale": 20,
  "effect": "Rain Neo",
  "localIp": "192.168.101.103",
  "device": "esp8266",
  "rColor": 3952730,
  "lColor": 4737104,
  "bDpth": 60,
  "mBri": 200,
  "freq": 10,
  "length": 30,
  "splashes": false,
  "clouds": true,
  "storm": false
}
homeassistant/light/firelamp-max2/set
{
  "state": "ON",
  "effect": "Rain Neo"
}

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
[... 387 lines cut ...]
<details>
<summary>Third stacktrace (captured one time, has different EXCCAUSE)</summary>

```
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
PC: 0x4024902c: sys_timeout_abs at core/timeouts.c line 214
EXCVADDR: 0x00003f2b

Decoding stack results
0x40249005: sys_timeout_abs at core/timeouts.c line 189
0x40249064: lwip_cyclic_timer at core/timeouts.c line 233
0x40249097: lwip_cyclic_timer at core/timeouts.c line 261
0x40248041: memp_free at core/memp.c line 447
0x402491f8: sys_check_timeouts at core/timeouts.c line 390
0x4024f708: mem_malloc at core/mem.c line 210
0x4024af06: pbuf_alloc_LWIP2 at core/pbuf.c line 284
```

</details>

I've tried to change effect to **Fire** and **Rain** and nothing crashed. Also, if I disable MQTT and change effect to **Rain Neo** via Web UI, nothing crashes too.

First stacktrace is pointing to `std::string::_Rep::_S_create(unsigned int, unsigned int, std::allocator  const&)` and what different in **Rain Neo** effect and other that Rain Neo effect has a way more fields in JSON (rColor, lColor, bDpth, mBri, freq, length, splashes, clouds, storm), so maybe there is a memory leak somewhere in MQTT callback? 

I'll try investigate this further tomorrow.

CODeRUS, 2021-01-14:
its not happening for me, please post your settings

maksimkurb, 2021-01-14:
<details>
<summary>config.json</summary>

```json
{
  "activeEffect": 0,
  "logInterval": 0,
  "matrix": {
    "width": 15,
    "height": 10,
    "segments": 1,
    "type": 13,
    "maxBrightness": 255,
    "currentLimit": 1500,
    "rotation": 3
  },
  "connection": {
    "mdns": "firelamp-max",
    "apName": "Fire Lamp (Max)",
    "ntpServer": "ru.pool.ntp.org",
    "ntpOffset": 14400,
    "hostname": "firelamp-max",
    "ssid": "***",
    "password": "*******************************"
  },
  "mqtt": {
    "host": "192.168.101.1",
    "port": 1883,
    "username": "*************",
    "password": "*********************************",
    "uniqueId": "firelamp-max",
    "name": "FireLamp (Max)",
    "model": "firelamp"
  },
  "button": {
    "pin": 4,
    "type": 1,
    "state": 0
  }
}
```

</details>

effects.json is from master branch (also tried without uploading it)

Sending this messages to **homeassistant/light/firelamp-max/set** topic:
```json
{
  "state": "ON",
  "effect": "Rain Neo",
  "speed": 70,
  "scale": 20,
  "bDpth": 60,
  "mBri": 200,
  "freq": 10,
  "length": 30,
  "splashes": false,
  "clouds": true,
  "storm": false
}
```

I've also noticed that ESP8266 frequently crash on [this line](https://github.com/CODeRUS/GyverLampCpp/blob/8532e40fca3212ac52f9b8b1e58f2563cecfb800/src/effects/basic/RainNeoEffect.cpp#L185) or on [this line](https://github.com/CODeRUS/GyverLampCpp/blob/8532e40fca3212ac52f9b8b1e58f2563cecfb800/src/MqttClient.cpp#L81), but it is not always true, so maybe it is just my luck.

CODeRUS, 2021-01-14:
ok, so, your case is happened by publishing same message multiple times?

maksimkurb, 2021-01-14:
Checked it, it happens when I change effect from **Rain Neo** to any other effect (even **Rain Neo** again). Sometimes crash can happen not after third message exactly, but after 5th or 9th.

Case 1: publishing **Rain Neo** message 3-10 times => crashed
```json
{
  "state": "ON",
  "effect": "Rain Neo"
}
```

Case 2: publishing **Fire** message 20 times => ok
```json
{
  "state": "ON",
  "effect": "Fire"
}
```
Case 3: publising messages **"Rain Neo" -> "Rain" -> "Rain Neo" -> "Rain"** => crashed
Case 4: publising messages **"Rain Neo" -> "Rain Neo" -> "Rain"** => crashed

maksimkurb, 2021-01-14:
I believe I have found an issue root 🥳, let me prepare PR

PR #48 (2021-01-14): [effects] Clear pointer after freeing memory in RainNeo effect
Fixes #47
closed 2021-01-14 by CODeRUS