CODeRUS/GyverLampCpp #53: matrix starts flickering once settings saved

alexandrbig, 2021-01-24:
**Using:** Wemos D1 mini, 16x16 matrix
**Issue:** after several seconds after start or effect change or button press matrix starts flickering. 

Connected the device to COM monitor to find any issues and right after 
```
Saving settings... Done!
Saving effects... Done!
```
it starts flickering.

What can cause such issues?

CODeRUS, 2021-01-24:
can you capture flickering on video?

alexandrbig, 2021-01-24:
https://user-images.githubusercontent.com/2073337/105628076-70030d00-5e43-11eb-9bfb-b1e493c84d33.MOV

alexandrbig, 2021-01-24:
https://user-images.githubusercontent.com/2073337/105628114-a17bd880-5e43-11eb-8c04-0b102edf41be.MOV

alexandrbig, 2021-01-24:
I am starting to record the video with lower speed and then changing speed to 100+ value. It is visible when settings are applied: the effect freezes for a second or less and then it starts flickering.

CODeRUS, 2021-01-24:
can't reproduce. you can try to disable dither in matrix settings.
also show your wiring, probably you powering your leds from 5v pin from board, or missing level shifter for led digital input pin.

alexandrbig, 2021-01-24:
+5v is taken directly from power supply then it goes to wemos +5 input.
Which level shifter do you mean? Resistor? On gyver’s guide it is mentioned that it is not needed if I am using matrix.

I tried to disable/enable dither option result is the same.

[image]

CODeRUS, 2021-01-24:
you need to add level shifter for DI led pin. ws2812 require 5v logic, esp8266 have 3v3 logic
you can get these ones: https://aliexpress.ru/item/32278943189.html

alexandrbig, 2021-01-24:
Ok I got your point. Thank you for the support and great port.

The only thing I cannot get why the Gyver's example works without any issue. 
Don't get me wrong, I am not trying to blame anyone, just want to make your solution work on the same hardware.

CODeRUS, 2021-01-24:
thats not a solution, just a suggestion. i've tested my d1 board without level shifter with following settings:
```
{
  "activeEffect": 1,
  "logInterval": 5000,
  "working": true,
  "matrix": {
    "pin": 2,
    "width": 16,
    "height": 16,
    "segments": 1,
    "type": 15,
    "maxBrightness": 255,
    "currentLimit": 1500,
    "rotation": 3,
    "dither": false
  },
  "connection": {
    "mdns": "devlampd1",
    "apName": "devlampd1",
    "apPassword": "",
    "ntpServer": "europe.pool.ntp.org",
    "ntpOffset": 10800,
    "hostname": "devlampd1",
    "ssid": "domhome",
    "bssid": "",
    "password": "omnomnom",
    "login": ""
  },
  "mqtt": {
    "host": "192.168.1.111",
    "port": 1883,
    "username": "username",
    "password": "mqtt",
    "uniqueId": "devlampd1",
    "model": "devlampd1",
    "name": "devlampd1"
  },
  "button": {
    "pin": 255,
    "type": 0,
    "state": 0
  },
  "spectrometer": {
    "active": false
  }
}
```
Tested with bouncing balls, sparkles and twinkles effects and can't get any issues or flickering. Are you using latest master?

alexandrbig, 2021-01-24:
I believe that it is some misconfiguration on my side. I tried to upload `settings.json` via react app and somehow it started to work. 🤦 
😄😄 😄  

Previously I changed settings with further fs building and uploading procedure.

I think uploading settings via web might save someone's time.

Thank you very much for the support!!!

closed 2021-01-29 by alexandrbig