CODeRUS/GyverLampCpp #60: After power loss the light stops functioning

justinmartin, 2021-02-19:
I can flash my NodeMCU and the matrix functions as expected.  I can change effects, brightness, and it connects to my MQTT server.  However, when I unplug the device and plug it into wall power, the grid turns green,  I can connect the webUI but no effects are applied.  When I reconnect it to my computer and look at the logs it says it can't find any effects.

I'm not sure how to troubleshoot more from here.

CODeRUS, 2021-02-19:
show your settings.json, probably you have wrong button config and lamp goes to configuration mode at boot

justinmartin, 2021-04-29:
I apologize for never getting back to this.  I finally made time tonight to try to and get this working again.  I didn't have a button hooked up, which seems to have been the issue.  After rereading the documentation several times, I disabled the button in my settings.json and the lamp is working perfectly.

Thank you.

closed 2021-04-29 by justinmartin