CODeRUS/GyverLampCpp #63: Cant connect to ESP

Doyle4, 2021-04-28:
Hi there,

I have been passed your links as I was struggling with the original all in Russian.

I have flashed the firmware to my ESP8266, I can see it in my wifi settings "Fire Lamp".

Im stuck here, I dont know how to connect to it to change effects etc.
Im use to WLED and accessing the board via 4.3.2.1,
Any help/advice would be amazing thanks :)

CODeRUS, 2021-04-28:
hello. please carefully read description (https://github.com/CODeRUS/GyverLampCpp#readme), you must flash firmware and filesystem images. if you flashed just firmware you can connect to device and open ip/update url and upload filesystem image here.

Doyle4, 2021-04-28:
Hi @CODeRUS 
Thanks for replying, This is the problem I am having, I am unable to connect to its IP to upload the filesystem.

I have tried 8.8.8.8, 8.8.9.108 (or if IP is different) as well as WLED's IP 4.3.2.1 but having no luck.

Tried since your comment, reflashed the board several times with just firmware and still not managing to connect to it via IP to update.

Doyle4, 2021-04-28:
Think I have solved it thanks, can access web page now via 8.8.8.8.

Problem Im having is when the esp reboots the LEDs set to solid green and have no function.

Doyle4, 2021-04-28:
Steps taken:
Flashed ESP firmware, rebooted esp, ticked "dont erase flash" and flashed esp-fs

Sorry about this, this is all new to me.

Doyle4, 2021-04-28:
Found a method to flashing it that works.

closed 2021-04-28 by Doyle4