CODeRUS/GyverLampCpp #64: Random ESP reboots

Doyle4, 2021-04-28:
Managed to get the esp working and seems to run great.

I have my 8X32 Matrix hooked up to a 5V PSU and esp powered by USB with esp and PSU sharing Ground.

Not sure why, but some effects when tweaking via web make the ESP reboot.
Same when changing text in scrolling text.

Any tips for this?

Compiled my own firmware in VSC and using the downloaded FS.bin

Also if possible to be answered, I know how to compile the main firmware using VSC, but unsure how to create my own FS, is this possible?

Thanks again, enjoying this firmware so far :) great work.

CODeRUS, 2021-04-28:
its common problem with network stack. do not perform rapid changes.

for compiling fs you can follow cli guide from readme, or find filesystem in tasks

Doyle4, 2021-04-28:
That will be why, I was changing often.

Thanks, will take a look soon. 👍🏻

closed 2021-04-28 by Doyle4