CODeRUS/GyverLampCpp #69: Compilation problem related to ESPAsyncE131-wled

Shor1, 2022-03-15:
Эта библиотека удалена из репозитория platformio? 
https://registry.platformio.org/search?q=ESPAsyncE131-wled
Если добавляю вручную из репозитория Aircookie, указав в platformio.ini ссылку на её git - возникают уже другие ошибки, связанные с аргументами некоторых фунций оттуда. Думал там устаревшая, не обновлявшаяся с 2019г версия библиотеки - удалил строку про эту библиотеку из platformio.ini, скопировал из ещё обновляющегося репозитория Aircookie/WLED файлы .сpp/.h этой библиотеки в папку lib проекта GyverLampCpp - всё равно ошибки аргументов.

CODeRUS, 2022-09-09:
what is the issue about?

Shor1, 2022-09-09:
At first it was thoroughly described by me in russisn language but as I saw that there was a response on some newly opened issue and no response to this - I deleted all explanation in russian, and left only heading in english.
The problem, if I remember right, is that platformio cant find such library in repo and thus cant compile.

CODeRUS, 2022-09-09:
the lib is inside lib folder: https://github.com/CODeRUS/GyverLampCpp/tree/master/lib/ESPAsyncE131-wled
It works out of box. In case of problems please post build output with errors.

closed 2022-09-09 by CODeRUS