CODeRUS/GyverLampCpp #71: Just to understand how it works

dragospascale, 2022-03-31:
In MyMatrix.cpp you have a function:
uint16_t MyMatrix::getPixelNumberXY(uint8_t x, uint8_t y)
{
    return static_cast<uint16_t>(XY(y, x));
}
Why invert coordinates ?
I'm trying to make a matrix from 8x8 tiles. 3 horizontal tiles and 2 vertical  It works well on the first line of tiles, but the clock in the second line of tiles does not look good. (the pixels seams to jump) .
I modified the test to draw a horizontal line and a vertical line on the second line of tiles  but the coordinates seems to have origin at 1x1 not 0x0 because drawLineXY(0, 8, 23, 8, Green) in fact draws a line from y = 7.
I cannot figure why. May you give me a hint where to look

CODeRUS, 2022-03-31:
tiling is setup using tile options: https://github.com/marcmerlin/Framebuffer_GFX/blob/master/Framebuffer_GFX.h#L57

dragospascale, 2022-04-01:
Yes I know how to setup for tile. I told you that it works somehow meaning that all the leds lightens. I saw that you changed XY function. It was just a mistake ?
The problem with led addressing still bother me and why the first line of tiles works ok and the others are not

dragospascale, 2022-04-03:
I'm closing I've made wrong wire connection between tiles

closed 2022-04-03 by dragospascale

CODeRUS, 2022-04-03:
thanks, it's good you sorted it out! :)