CODeRUS/GyverLampCpp #72: Угловой светодиод мигает разными цветами при вай-фай подключении.

Shor1, 2022-07-23:
В январском релизе такого нет, а в крайнем релизе один из угловых диодов становится каким-то индикатором активности соединения, когда подключаешься к точке доступа Fire Lamp.
[image]
Платформа esp8266 (esp01s), пин матрицы - 2, пин кнопки - 255, rotation - 1

zerkalica, 2022-09-09:
https://github.com/FastLED/FastLED/issues/1322

try downgrade platform in `platformio.ini`:

```
[esp8266]
...
platform = espressif8266@2.6.3`
```

Shor1, 2022-09-09:
Due to issue #69 I can't compile it, but use author's binaries, so it is up to him to check if this will work.

Shor1, 2022-09-13:
Well partial success: compiled the same release from 28.06.22 with core 2.6.3, flashed it and this flickering doesnt occur. Flashed author's binaries - it showed up. Compiled same version with default core settings - it compiled successfully but flashed binary doesn't run properly (Firelamp AP doesnt appear on power on, tried flashing twice) - so it was partially clear experiment: cant check binaries compiled on the same machine/environment and flashed on the same board side by side but at least solution with forcing esp core to version 2.6.3 is working.

CODeRUS, 2022-12-18:
Fixed with b03352e08c0d7e5372c68e2c4424165d7e8513bc

closed 2022-12-18 by CODeRUS