CODeRUS/GyverLampCpp #80: ESPhome config for Home Assistang.

cracrama, 2023-06-10:
Hi
Many thanks for great work. 
I would like to ask if there is available config for ESPhome ?
Is it possible to create one for ESPhome users?
Please advise. 
Kind regards.

CODeRUS, 2023-06-10:
@cracrama this is not for esphome, it discovers automatically via mqtt integration

closed 2023-06-10 by CODeRUS

cracrama, 2023-06-10:
Many thanks for response. I will try upload this firmware then. 
Best regards

CODeRUS, 2023-06-10:
@cracrama make sure to read description for the steps and configuration

cracrama, 2023-06-11:
Many thanks but i'm quite new in platformio and i'm bit lost . Do you have maybe link to some video which explains step by step process and which files should i copy? When I'm copying and extracting "data.zip" - seems like there is file missing : src/EffectsManager.cpp?
Please advise

CODeRUS, 2023-06-11:
EffectsManager is part of source code, not data folder :)