CODeRUS/GyverLampCpp #81: Unavailable in Home Assistant

wardy277, 2023-08-24:
I am not sure if this is a home assistant or mqtt.org server update change. But the lamp, which has been working fine has suddenly become unavailable in home assistant 

I can still see and interact with the lamp via the webui. I can see that the device is registered in MQTT. I have changed the settings with a new name, uploaded to the device and can see the new device appear in MQTT as well as home assistant, but with the same issue of being unavailable. I have even reflashed both firmware.bin and fs.bin with a latest build

Does the code need updating with patches to support a new version of home assistant/mqtt?

Settings:
```
{
  "activeEffect": 0,
  "logInterval": 0,
  "matrix": {
    "pin": 4,
    "width": 16,
    "height": 16,
    "segments": 1,
    "type": 15,
    "maxBrightness": 255,
    "currentLimit": 5000,
    "rotation": 3,
    "dither": true,
    "order": "grb"
  },
  "connection": {
    "mdns": "firelamp",
    "apName": "Fire Lamp",
    "apPassword": "****",
    "ntpServer": "uk.pool.ntp.org",
    "ntpOffset": 0,
    "hostname": "firelamp",
    "ssid": "****",
    "bssid": "",
    "password": "****",
    "login": ""
  },
  "mqtt": {
    "host": "****",
    "port": 1883,
    "username": "****",
    "password": "****",
    "uniqueId": "firelamp",
    "name": "firelamp",
    "model": "firelamp"
  },
  "button": {
    "pin": 255,
    "type": 1,
    "state": 0
  }
}
```
MQTT Diagnostics
[mqtt-514072268cec2d4aa391b17e1c927ea4-firelamp-77831687de9b8825071da4cac8df4500.json.txt](https://github.com/CODeRUS/GyverLampCpp/files/12426751/mqtt-514072268cec2d4aa391b17e1c927ea4-firelamp-77831687de9b8825071da4cac8df4500.json.txt)

CODeRUS, 2023-08-24:
check serial messages on startup to see mqtt conenction logs

wardy277, 2023-08-24:
Does anything in the serial log show any idea whats going on? The `homeassistant/light/firelamp/state` lines near the bottom is where i used the webui to turn the lamp on and off

```
Happy debugging!
Flash ide  size: 4194304 bytes
Flash ide speed: 40000000 Hz
Flash ide mode:  DIO
Flash real id:   001640D8
Flash real size: 4194304 bytes

Flash Chip configuration ok.
Sketch size: 639936
Sketch free: 2502656
1_FreeHeap: 39200
Initializing Settings
FLASHFS Settings file exists: true
FLASHFS Settings save file exists: true
FLASHFS Settings file size: 632
reading settings.json.save
{"activeEffect":45,"logInterval":0,"working":false,"matrix":{"pin":4,"width":16,"height":16,"segments":1,"type":15,"maxBrightness":255,"currentLimit":5000,"rotation":3,"dither":true,"order":"grb"},"connection":{"mdns":"firelamp","apName":"Fire Lamp","apPassword":"**REDACTED**","ntpServer":"uk.pool.ntp.org","ntpOffset":0,"hostname":"firelamp","ssid":"**REDACTED**","bssid":"","password":"**REDACTED**","login":""},"mqtt":{"host":"**REDACTED**","port":1883,"username":"**REDACTED**","password":"**REDACTED**","uniqueId":"firelamp","model":"firelamp","name":"firelamp"},"button":{"pin":255,"type":1,"state":0},"spectrometer":{"active":false}}
Copying file from /settings.json.save to /settings.json
Button pin: 255
Initializing EffectsManager
FLASHFS Effects file exists: true
FLASHFS Effects save file exists: true
FLASHFS Effects file size: 4237
reading effects.json
[{"i":"Fire","n":"Fire","s":37,"l":1,"b":80},{"i":"Sparkles","n":"Sparkles","s":121,"l":12,"b":255},{"i":"VerticalRainbow","n":"Vertical rainbow","s":100,"l":18,"b":80},{"i":"HorizontalRainbow","n":"Horizontal rainbow","s":100,"l":18,"b":80},{"i":"Colors","n":"Color change","s":112,"l":1,"b":80},{"i":"CloudNoise","n":"Clouds 3D","s":1,"l":5,"b":80},{"i":"LavaNoise","n":"Lava 3D","s":1,"l":5,"b":80},{"i":"PlasmaNoise","n":"Plasma 3D","s":1,"l":5,"b":80},{"i":"RainbowNoise","n":"Rainbow 3D","s":1,"l":5,"b":80},{"i":"RainbowStripeNoise","n":"Rainbow stripe 3D","s":1,"l":5,"b":80},{"i":"ZebraNoise","n":"Zebra 3D","s":1,"l":5,"b":80},{"i":"ForestNoise","n":"Forest 3D","s":1,"l":5,"b":80},{"i":"OceanNoise","n":"Ocean 3D","s":1,"l":5,"b":80},{"i":"Color","n":"Single color","s":10,"l":1,"b":80,"color":{"r":255,"g":180,"b":200}},{"i":"Snow","n":"Snow","s":255,"l":33,"b":80,"color":{"r":255,"g":255,"b":255}},{"i":"Matrix","n":"Matrix","s":255,"l":25,"b":80},{"i":"Lighters","n":"Lighters","s":127,"l":10,"b":80},{"i":"Clock","n":"Clock vertical","s":250,"l":1,"b":80,"hColor":{"r":255,"g":255,"b":255},"mColor":{"r":255,"g":255,"b":255}},{"i":"ClockHorizontal1","n":"Clock horizontal colored","s":250,"l":1,"b":80,"hColor":{"r":255,"g":255,"b":255},"mColor":{"r":255,"g":255,"b":255}},{"i":"ClockHorizontal2","n":"Clock horizontal","s":250,"l":1,"b":80,"hColor":{"r":255,"g":255,"b":255},"mColor":{"r":255,"g":255,"b":255}},{"i":"ClockHorizontal3","n":"Clock horizontal single","s":250,"l":1,"b":80,"hColor":{"r":255,"g":255,"b":255},"mColor":{"r":255,"g":255,"b":255}},{"i":"Starfall","n":"Starfall","s":80,"l":100,"b":80,"step":100,"sat":150},{"i":"DiagonalRainbow","n":"DiagonalRainbow","s":100,"l":20,"b":80},{"i":"Waterfall","n":"Waterfall","s":100,"l":20,"b":80,"cooling":32,"sparkling":80},{"i":"TwirlRainbow","n":"Twirl Rainbow","s":100,"l":20,"b":80,"step":8,"twirl":300},{"i":"PulseCircles","n":"Pulsing circles","s":100,"l":20,"b":80,"m":1},{"i":"Animation","n":"Aqarium","s":100,"l":20,"b":80},{"i":"Storm","n":"Storm","s":100,"l":20,"b":80,"colored":false,"density":32},{"i":"Matrix2","n":"Matrix 2","s":100,"l":20,"b":80},{"i":"TrackingLighters","n":"Tracking Lighters","s":100,"l":20,"b":80,"tails":1},{"i":"LightBalls","n":"Light Balls","s":100,"l":20,"b":80,"size":1},{"i":"MovingCube","n":"Moving Cube","s":100,"l":20,"b":80,"rnd":true},{"i":"WhiteColor","n":"White Color","s":100,"l":20,"b":80},{"i":"PulsingComet","n":"Pulsing Comet","s":100,"l":20,"b":80},{"i":"DoubleComets","n":"Double Comet","s":100,"l":20,"b":80},{"i":"TripleComets","n":"Triple Comet","s":100,"l":20,"b":80},{"i":"RainbowComet","n":"Rainbow Comet","s":100,"l":20,"b":80},{"i":"ColorComet","n":"Color Comet","s":100,"l":20,"b":80},{"i":"MovingFlame","n":"Moving Flame","s":100,"l":20,"b":80},{"i":"FractorialFire","n":"Fractorial Fire","s":100,"l":20,"b":80},{"i":"RainbowKite","n":"Rainbow Kite","s":100,"l":20,"b":80},{"i":"BouncingBalls","n":"Bouncing Balls","s":100,"l":20,"b":80,"colored":false},{"i":"Spiral","n":"Spiral","s":29,"l":20,"b":80,"count":1},{"i":"MetaBalls","n":"Meta Balls","s":100,"l":20,"b":80},{"i":"Sinusoid","n":"Sinusoid","s":100,"l":20,"b":80},{"i":"WaterfallPalette","n":"Waterfall Pallete","s":100,"l":20,"b":80,"sparkling":80},{"i":"Rain","n":"Rain","s":100,"l":20,"b":80},{"i":"Prismata","n":"Prismata","s":100,"l":20,"b":80},{"i":"Flock","n":"Flock","s":100,"l":20,"b":80,"predator":false},{"i":"Whirl","n":"Whirl","s":1,"l":26,"b":80,"colored":true,"speed":1,"scale":26},{"i":"Wave","n":"Wave","s":100,"l":20,"b":80},{"i":"Fire12","n":"Fire12","s":100,"l":20,"b":80,"cooling":70,"sparking":130,"smoothing":80},{"i":"Fire18","n":"Fire18","s":100,"l":20,"b":80},{"i":"RainNeo","n":"Rain Neo","s":100,"l":20,"b":80,"rColor":{"r":255,"g":255,"b":255},"lColor":{"r":255,"g":255,"b":255},"bDpth":60,"mBri":200,"freq":50,"length":30,"splashes":false,"clouds":false,"storm":false},{"i":"Twinkles","n":"Twinkles","s":4,"l":20,"b":80,"espeed":4,"mult":6},{"i":"DMX","n":"e131 or art-net","s":0,"l":20,"b":81,"uni":1,"port":6454," [... cut ...]
[... 162 lines cut ...]
<< {"event":"WORKING","data":false}
working: false
Sending state
homeassistant/light/firelamp/state
{
  "state": "OFF",
  "brightness": 80,
  "speed": 100,
  "scale": 20,
  "effect": "Waterfall Pallete",
  "localIp": "**REDACTED**",
  "webui": "http://**REDACTED**",
  "device": "esp8266",
  "sparkling": 80
}
Sending state to ws clients
{
  "activeEffect": 45,
  "working": false
}
Saving settings... Done!
Saving effects... Done!
FrameBuffer::GFX 95fps
FrameBuffer::GFX 101fps
```

CODeRUS, 2023-08-24:
okay, it says mqtt is connected. is lamp the only mqtt entity on your home assistant? can you check if you can receive any mqtt messages at all?

wardy277, 2023-08-30:
Sorry, for the delay. I have been playing with settings and mqtt, and what i have found is odd.

I have found an mqtt explorer which allows me to see whats going on in the world of mqtt. I can see that the lamp has 3 nodes config, state and available. I noticed that available has a value of false. I simply published `true` to `homeassistant/light/firelamp/available` and yo and behold, the lamp was available in home assistant.

I also tested this by publishing the available=true from within home assistant. No idea where the issue lies, with mqtt, home assistant or the lamp itself